espruino/BangleApps #2502: Feature request: low power mode

thyttan, 2023-01-14:
When the battery goes below a low battery level, enter low power mode only chowing the time, not allowing any other features and no backlight. I imagine this would be baked into the firmware, but I guess an app could accomplish most of these things as well. Inspiration for the request: https://youtu.be/VXSUQSpBYTo?t=327

gfwilliams, 2023-01-16:
I seem to remember this being requested before as well, but given the relatively long battery life normally it's probably not super high priority.

There are still some things (like alarms) that it's expected would function, but I bet this could be made into an app (maybe with some changes to the `boot` app to restrict what gets loaded).

myxor, 2023-01-23:
It could also do some changes to the settings, e.g.:

- LCD brightness
- LCD timeout
- Disable auto heart rate measurement in health app
- Overwrite setGPSPower() function?
- ...

hughbarney, 2023-02-07:
* turn wake on twist off

pavelmachek, 2023-03-14:
I would not mind having something like this. I left GPS logging running for too long, and result was crashed watch and lost time. It would be more friendly to go to some kind of "safe mode" (only showing time, GPS off) when battery goes to 5%.

gfwilliams, 2023-03-15:
Thanks - while I'm not a great fan of a general low power mode as I mentioned above, I can totally imagine cases like yours where GPS gets left on, in which case it would be really handy to have the watch automatically turn it off if the battery got too low.

I could be wrong here but it feels like really GPS is the only 'big power draw' thing that might get turned on that'd flatten the watch battery in less than a day (apart from the backlight, but it'd be rare that anyone was able to turn that on and leave it on).

Maybe turning GPS off automatically should be built into the watch itself

> and result was crashed watch

What actually happened? You mean the watch just turned off? Or there was some other issue?

bobrippling, 2023-04-07:
I've [pushed up an app](https://github.com/espruino/BangleApps/pull/2684) that does a fair chunk of this - if anyone fancies giving it a go, let me know what you think / PRs also welcome.

PR #2684 (2023-04-20): Drained: Low power mode app
This adds an app which (if installed) will automatically load at 5% battery and attempt to put the watch into a low power mode.

(The `setUI` and `removeAllListeners` changes from espruino/Espruino#2348 are included for the build)

Checklist:
- [X] Loads on low battery
- [X] Power cycling restores `drained` app
- [X] Charging reaching a threshold restores original watch app
  - [X] Regardless of whether charging occurs before or while `drained` is running
- [X] Permit user cancellation of low power mode

Closes #2502
closed 2023-04-20 by gfwilliams