espruino/BangleApps #2513: [app/inspire] Not available in the store

trufae, 2023-01-19:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

My PR was merged last week but i still dont see it listed in the development page of the bangleapps site. did i missed anything?

gfwilliams, 2023-01-19:
No - the update from the development app loader at https://espruino.github.io/BangleApps/ to the main one is kicked off manually by me every so often, and I'm a bit busy recently so didn't have a chance to do it.

I've just started it now so your app should appear in maybe 30 minutes

closed 2023-01-19 by gfwilliams

trufae, 2023-01-20:
Uhm , sorry but its still not there

nxdefiant, 2023-01-20:
It does show up for me for BangleJS1, do you expect it to be installable for BangleJS2? because it is not configured to do so
https://github.com/espruino/BangleApps/blob/master/apps/inspire/metadata.json#L10