espruino/BangleApps #2525: [messages] LCD backlight turns on after auto-exiting, not when messages arrive

digitalcircuit, 2023-01-24:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

### Describe the bug
With Message GUI configured to auto-open, the LCD backlight doesn't turn on when showing the incoming message.

Instead, the LCD backlight turns on after the Message GUI app closes, then turns off after the normal LCD backlight timeout.

### Expected behavior
The LCD backlight turns on when the incoming message is shown via auto-open, allowing reading the message without having to unlock the watch.

### Steps to reproduce
1.  Set up the latest development versions of the Bangle.js 2 apps, including Android support
     * I tested with Anton Clock as default, also happens with Circles Clock
     * I tried setting LCD backlight brightness to 1 (instead of 0.2), still happened
2.  Connect and pair with Gadgetbridge, ensuring Messages is set to auto-open (not disabled) but *not* auto-unlock (default)
3.  Return to home clock screen
4.  Wait for LCD backlight to turn off
5.  Send test notification via Gadgetbridge, or wait for a normal notification

### Installed apps

antonclk (0.11), widbat (0.09), widbt (0.08), mylocation (0.09), gpsautotime (0.04), locale (0.17), widalarmeta (0.01), quicklaunch (0.08), multitimer (0.03), boot (0.55), launch (0.20), about (0.14), owmweather (0.02), messages (0.58), clkinfosunrise (0.03), kbswipe (0.07), widlock (0.08), widmessages (0.04), notify (0.12), setting (0.56), alarm (0.37), messageicons (0.04), messagegui (0.64), android (0.21), agpsdata (0.06), health (0.22), weather (0.24), circlesclock (0.24), agenda (0.12), sched (0.21)

gfwilliams, 2023-01-24:
Thanks - have you got `Settings -> Apps -> Messages -> Unlock Watch` enabled?

digitalcircuit, 2023-01-24:
I do not have `Unlock Watch` enabled (I think that's the default).

Enabling it works around the issue, so perhaps there's something about `Bangle.setLCDPower(1);` not applying while the watch is locked until there's an app switch/call to `load()` while locked?

Changes for `Settings → Apps → Messages`:
* Messages
   * Vibrate: `::` (from default `:`)
   * Vibrate for calls: `:::` (from default `:`)
   * Repeat: Off (from default `4s`)
   * Uncheck `Flash Icon` (from default checked)
   * Check `Quiet mode disables auto open` (from default unchecked)

(If desired, I can share my entire list of changes.  Ever since resetting my Android phone a few years ago, I've started writing down *every* configuration change…)

gfwilliams, 2023-01-24:
As far as I understand it, the idea of 'unlock watch' was that it did light the backlight up...

Do you see `Bangle.setLCDPower(1);` in the code somewhere and it's not having an effect then? Or you just expected that the app should light up without you having to set that setting?

digitalcircuit, 2023-01-24:
I apologize, I got confused between `messagelist` (a separate app) and `messagegui`.  I also completely misremembered how the watch behaved.  Reading the code, including before the `messages` → `messagegui/widmessages/messages` split, the watch is functioning as it should.

I guess I might've assumed that the watch was supposed to be turning on the screen when showing the message because it was turning on the screen *after* showing the message.  But I think that's a [separate, already filed bug report](https://github.com/espruino/BangleApps/issues/1873 ).

Optionally, this could be a feature request to have it turn on the screen without unlocking, to allow reading without risking accidental taps/swipes ([moving the check for `unlockWatch` after and inside the `quietMode` check](https://github.com/espruino/BangleApps/blob/1c7eb92ad5e2672154565b6ac0c487f50aa0a366/apps/messagegui/lib.js#L73-L76 )).

However, I don't know if folks prefer the current behavior.

digitalcircuit, 2023-02-14:
I may try filing a feature request for turning the screen on without unlocking, or I may just opt for the unlock watch option.  And the issue that the screen turns on for me *after* the message is dismissed (on app switch) is another known bug.

Either way, this bug report as it stands is invalid, so I'll close it to help clean up the clutter.

closed 2023-02-14 by digitalcircuit

thyttan, 2024-05-09:
@digitalcircuit if you're still around, you may find [Message wake on face up](https://espruino.github.io/BangleApps/?q=msgwakefup) useful.