espruino/BangleApps #2538: Feature Request:  Debug option to log to log.txt only

hughbarney, 2023-01-29:
As discussed on this thread:
https://forum.espruino.com/conversations/383367/#comment16849198

The Settings/Utils/Debug settimgs has 3 settings:  Hide, Show, Log

The Log option currently displays as well.  I would rather just log rather than interrupt the display with other debug msgs.
I think this would be clearer if it was done as 2 check boxes.
Display [] and Log [] this gives you 4 options.

Also the ability to always have exceptions sent to log.txt

gfwilliams, 2023-01-30:
Yes - good idea. PRs welcome :)

hughbarney, 2023-01-31:
Am I right in that I would only have to change the way bootupdate.js sets up the logging and console ?

That way I could try just using values for settings.log as 0,1,2,3
Maybe later fix up the settings UI to have the 2 check boxes and set the value of settings.log accordingly

Settings.log values

               Existing Label         New Label
                -------------------          ---------------
"log":0,  Hide,                        off
"log":1,  Show                        display
"log":2   Log                           log    (currently logs and displays)
"log":3                                     both

bobrippling, 2023-11-15:
> I would only have to change the way bootupdate.js sets up the logging and console ?

I can only find mention of the loopbacks in `bootupdate.js`, but I couldn't say for certain. But yes - other than that you seem along the right lines

closed 2023-11-16 by gfwilliams