espruino/BangleApps #2562: Exception "Uncaught Error: Cannot read property 'reset' of undefined"

myxor, 2023-02-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

I am not quiet sure where this error happens but it happens on my side when i try to edit an existing alarm via the alarms app:

```
Uncaught Error: Cannot read property 'reset' of undefined
 at line 2 col 171
...elete Bangle.uiRemove;a()}g.reset();if(c){if("updown"==c){va...
                              ^
in function called from line 2 col 291
...:c.x,y:c.y+f-b.y-a*d.h}))}});var b=Bangle.appRect,h=0|d.scro...
                              ^
at line 2 col 291
...:c.x,y:c.y+f-b.y-a*d.h}))}});var b=Bangle.appRect,h=0|d.scro...
                              ^
in function called from line 1 col 28
m.scroller=E.showScroller(r)
                           ^
at line 1 col 28
m.scroller=E.showScroller(r)
                           ^
in function "q" called from line 7 col 422
....scroller.drawItem(a)}}};q();return m
                              ^
in function called from line 1 col 2677 in alarm.app.js
...10,a,d,g))})}),E.showMenu(e)
                              ^
at line 1 col 2677 in alarm.app.js
...10,a,d,g))})}),E.showMenu(e)
```

i currently got no time to investigate this but maybe someone else can have a look :)

dashavoo, 2023-02-06:
Try deselecting all the minification options etc. in the app loader and then reinstall.

nxdefiant, 2023-02-07:
Neither alarm nor sched.js do a g.reset(). Can you please post your installed apps with versions?

gfwilliams, 2023-02-07:
I think the code that's causing the problem will be in E.showScroller by the look of it? But the issue is really that *something* has overwritten/deleted the global variable `g` which is used for the graphics.

I just tried and I can't reproduce here by editing a pre-existing alarm - maybe you could post some more specific steps? Also see if it happens in the emulator or with a fresh install?

It might also be worth logging in with the IDE and checking `g`. Maybe keep checking while you're doing different things and find out at which point it gets altered. That might help?

You could also try pasting this code into the REPL:

```
var oldG = g;
delete g;
Object.defineProperty(global, "g", { get : () => oldG, set : () => {
  throw new Error("Don't set g");
}});
```

It should then throw an exception if something tries to overwrite `g` which should give you a stack trace (sometimes the widget swipeOn code does it, but that is fine and shouldn't be an issue here)

myxor, 2023-02-07:
> Try deselecting all the minification options etc. in the app loader and then reinstall.

This makes the difference. When installing with minification the error occurs, when installing without minification it works as expected.

> It might also be worth logging in with the IDE and checking g. Maybe keep checking while you're doing different things and find out at which point it gets altered. That might help?

No difference in the `g` object before and after opening alarms and triggering the error.

gfwilliams, 2023-02-07:
Ok, great - thanks! What happens if you toggle the minification option and then install *just* the alarm app?

If we're sure that minification of just one app changes things then at least that will help to narrow down where the problem lies.

myxor, 2023-02-07:
> Ok, great - thanks! What happens if you toggle the minification option and then install _just_ the alarm app?

This is what i did. If minification enabled it crashes if disabled it works.

gfwilliams, 2023-02-07:
Thanks! We could attempt to fix the alarm app with some kind of hack, but I there will be others. I think the only real fix to to try and figure out how to tell esprima/escodegen not to use certain variables.

I'm tempted to remove the minification option again? Or at the very least change the text `BETA` to something a lot stronger that explains that many apps will likely break.

The actual performance improvements are pretty minimal, and the downside of a bunch of otherwise good apps failing seems pretty huge.

bobrippling, 2023-04-24:
I've been running into the same problem - when an alarm triggers, I would see my watch freeze on loading, and when attached to an IDE, I saw the below backtrace (this exception halting the load).

In each case, `g`'s properties weren't defined (whereas `g` itself seems ok - the errors talk about `g.getWidth` or `g.theme` being `undefined`). This lines up with esprima possibly using `g` as a short variable.

Oddly, when I check `g` after the backtraces, these properties are present, but having reinstalled all apps (without the minification option), the issue is gone.

<summary>
Backtrace
<details>

```
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v17 (c) 2021 G.Williams
hwid_a_battery_widget.wid.js Error: Error: Cannot read property 'dark' of undefined
Uncaught Error: Function "getWidth" not found!
 at line 1 col 28
if(global.WIDGETS){var d=g.getWidth(),e=g.getHeight(),c={tl:...
                           ^
in function called from line 1 col 2222 in sched.js
...dgets(),Bangle.drawWidgets();let d=b.buzzCount;if(E.showPrompt(c...
                                  ^
at line 1 col 2222 in sched.js
...dgets(),Bangle.drawWidgets();let d=b.buzzCount;if(E.showPrompt(c...
                                  ^
in function "showAlarm" called from line 1 col 4216 in sched.js
....length?showAlarm(active[0]):setTimeout(load,100)
                              ^
Uncaught Error: Function "reset" not found!
 at line 1 col 256 in widdevst.wid.js
...age/e.total;var a=this.x,c=this.y;g.reset(),g.clearRect(a,c,a+21,c+23...
                                       ^
in function called from line 24 col 1960 in .boot0
....apply(undefined,arguments.slice(2));let end=Date.now()-start;let f=a.toSt...
                                            ^
in function called from system
Uncaught Error: Function "reset" not found!
 at line 1 col 256 in widdevst.wid.js
...age/e.total;var a=this.x,c=this.y;g.reset(),g.clearRect(a,c,a+21,c+23...
                                       ^
in function called from line 24 col 1960 in .boot0
....apply(undefined,arguments.slice(2));let end=Date.now()-start;let f=a.toSt...
                                            ^
in function called from system
```

</details></summary>

gfwilliams, 2023-04-24:
Thanks. I'm honestly not sure what to suggest though - increasingly I'm thinking I should just remove the option (as I did previously).

The minification just isn't reliable and doesn't give a good enough benefit for the times it works

bobrippling, 2023-04-24:
Yeah, I think I'd agree with you - we could look into tweaking the minifier for specific variables, but I expect that would break if we added more in future. I could sort a PR to remove the option?

bobrippling, 2023-04-25:
Sounds good - done (happy to tweak the look & feel)

bobrippling, 2023-11-15:
Going to close this as the problem's solved by #2709

closed 2023-11-15 by bobrippling