espruino/BangleApps #2575: Time goes forward by one hour when phone is not connected

amunizp, 2023-02-12:
When my phone looses connection the watch seems to change timezone by one hour. When I check the bangle.js app (gadget bridge) and go to "more..."  it is set so that "always update time when we connect". 

What I don't understand is why if I choose to disconnect why does it find itself going forward one hour?

amunizp, 2023-02-20:
Ok. I think I might have found the issue and it might be to do with my lack of knowledge on how blue tooth works. 

It does not seem to happen immediately after it disconnects. I did notice that it did not go back the the phone time until I disconnected my blue tooth headphones. I don't do anything on the watch to get it to disconnect. Sadly I do not see when the time changes. I am normally in the London UK time zone but some weeks ago I did change timezone. But that is about the time that I started to use Bluetooth headphones as well.

Firmware is up to date. I did do rewrite settings on the watch just now (before realising that my Bluetooth headphones were on). And the time did not change. But it did change back to the correct time when I disconnected the blue tooth headset.

I have now connected the Bluetooth headset again. And I used the bangle gadget bridge app to remove an app. It did it. But moved the time on watch forward by one hour. Tried rewrite settings and still is one hour forward. I then put bluetooth earbudd into the box and after a minute or so the time went back to the real time.

Hope this helps?

amunizp, 2023-02-20:
Ok. The time went back forward again. Not sure why. I tried connecting and disconnecting again but it did not do anything. I then took my Bluetooth head phones out and it connected immediately.

Let me know what tests would give better information to what is going on.

gfwilliams, 2023-02-22:
Ok, thanks. The Bluetooth headphones should really have no effect on other bluetooth devices like the Bangle so I'm not sure that's it.

Please can you try going into Gadgetbridge  (on your phone), the main settings page, and disabling `Sync time`.

I wonder whether what's happened is the web browser (including in the Bangle app store within Gadgetbridge) and Gadgetbridge itself somehow have a different idea about what the current Timezone is, and depending on what you used last, things get confused.

amunizp, 2023-02-22:
OK  made sure I was on the right time and I have disabled the time sync thing. Would like to have it sync in day light saving hours though. 

But still puzzled about the Bluetooth headphones. It seemed fairly consistent. 

Will keep this posted. Thank you!

amunizp, 2023-02-22:
Nope. It just went forward again. Just as I connected blue tooth headphones. Puzzling. Is there a log I can create?

amunizp, 2023-02-23:
I am using the f-droid bangle gadget bridge app. 

I see under *More...*

Yes *Always update time when we connect* was enabled I have disabled it now and left the *sync time* on.

I noticed there is a button under More> utilities,>  that says update time. That set my time on sync no problem

amunizp, 2023-02-27:
OK. More experiments. F-droid 0.72 version. Just noticed there was an update to 0.73 4 days ago.

Bangle.js Gadget bridge> settings > date & Time > Sync time > ON

Bangle.js Gadget Bridge> bangle.js > apps manager> More... > Settings > always update time when we connect > ON

The time on the watch is one hour ahead.

So I go to:
Bangle.js Gadget Bridge> bangle.js > apps manager> More... >Utilities>  set bangle.js Time

I get the green banner at the bottom saying "time set successfully" . But the time on watch is still the same.

To set the time in sync I go to:
Bangle.Js gadgetbridge> bangle.js > three dots> disconnect 

Wait until disconnected message comes on the bottom

Repeat to connect. The time is set on sync.

But I then go to bed andwake up and the watch is once again set to Spain time (one hour ahead).

gfwilliams, 2023-02-27:
Ok, so Spain time (one hour ahead) is not what it should be?

Also, I just wanted to check you don't have the `Daylight Saving` app installed? I guess that could be messing with things.

Please can you try doing a long-press of the Button on Bangle.js (~3 seconds) after you've tried setting time the different ways? That will reboot the clock so any changes will definitely take effect - it may be that things are getting changed and then not showing on the screen immediately and then it's getting hard to see exactly which thing is causing the problem.

amunizp, 2023-02-27:
Sorry for the confusion: it is one hour ahead from the real time which is what my phone says.

I do not have the daylight saving app installed.

Bangle.js Gadget Bridge> bangle.js > apps manager> More... >Utilities> set bangle.js Time

I get the green banner at the bottom saying "time set successfully" . But the time on watch is still the same.

I then hold the button for 3 seconds. The bangle.js rebooted and the time set to the same time as mobile.

After some time (less than hour?) The time on the watch moved one hour forward.

gfwilliams, 2023-02-28:
Thanks! So it looks like:

> Bangle.js Gadget Bridge> bangle.js > apps manager> More... >Utilities> set bangle.js Time

This sets the time correctly. I know it doesn't show immediately but after the long press it'll show the right time.

**What happens if you disconnect from Gadgetbridge after this, long-press the button just to be sure the time is still ok, and leave the Bangle without connecting it to Gadgetbridge?**

This would give a much clearer idea of whether something in the watch is messing things up, or if Gadgetbridge is automatically setting your watch's time to the wrong thing...

amunizp, 2023-02-28:
OK so the time was holding OK for today I thought that because I upgraded the app to 0.73 it might have been fixed.

so I uninstalled parrot. That caused the time on my watch to be one hour ahead from my phone.
I then did the set bangle.Js time, and held the button as mentioned and after a while the time came in sync with my phone again.

I then disconnected the app on the phone bangle.js gadget bridge. And long pressed the button on the watch.

The watch is yet again one hour ahead from what my phone tells me.

I connected the bangle.js gadgetbridge and the watch is now back on sync.

So the watch is to blame? Could it be a setting or an app installed?

According to gadget bridge I have:
Bluetooth widget
Bluetooth ID widget
Welcome
Sleeplog
Alarms widget
Launcher
Anton clock
About
Messages
CPR assist
Image background
Notifications (default)
Lock widget 
Message widget
Weather
Health tracking
LCARS (more on this one later)
Alarms and timer
Scheduler
Android integration
Bootloader
Settings
Message icons
Battery level widget
Message UI

gfwilliams, 2023-03-01:
**Am I right in saying that whenever you do the ~3 sec press to reset the watch, the time ends up wrong? And it only sorts it out at some point later when you're connected to Gadgetbridge?**

I'm struggling to figure out what's causing it. It feels like it's because the watch isn't saving the timezone correctly.

If you completely disconnect from Gadgetbridge and connect to the Web IDE (espruino.com/ide) on a computer, then do the ~3 second button press, do you see any error messages?

> so I uninstalled parrot. That caused the time on my watch to be one hour ahead from my phone.

What did you uninstall Parrot with? The App Loader within Gadgetbridge?

>  after a while the time came in sync with my phone again.

After roughly how long? You mean right after doing the ~3 sec press of the button, the time was still wrong, and it only changed at some point (minutes?) later *while connected to gadgetbridge*?

Please could you do a backup of your watch (App Loader -> More... -> Backup) and send it to me? It will back up all the files on there so if there's anything sensitive feel free to send it to gw@pur3.co.uk instead. But it might help me to see what's going wrong.

amunizp, 2023-03-20:
OK Sorry for the delay. 
First about connecting to the IDE. The time currently is one hour ahead. 
[image]
I disconnected from android phone gadget bridge and reconnect using laptop using chromium. I later held down the button until the bar loaded and let go. I got the message stating that I disconnected. I then requested to get it connected.

amunizp, 2023-03-20:
I am now connected via chromium browser using https://banglejs.com/apps/?q=open 
I click set bangle.js time in the More... menu. and after a few seconds the time is now in the correct time. 

I then uninstall the welcome up. The watch asks me to hold down the button to reload and the time went forward one hour. i have not lost connection with the chromium page. 

I have emailed you the zip of the back up.

gfwilliams, 2023-03-20:
Ok, thanks - that is extremely helpful and really narrows it down! 

I can't see the email has arrived yet - I guess it may have bounced back from my email account for some reason. Maybe you could upload it somewhere like Google Drive and share it with me that way?

amunizp, 2023-03-20:
A: so I uninstalled parrot. That caused the time on my watch to be one hour ahead from my phone.

    Q: What did you uninstall Parrot with? The App Loader within Gadgetbridge?

Yes I uninstalled it with Gadget bridge but the same thing I have just demonstrated by doing the same with the welcome app using the web app loader. 

[image]

gfwilliams, 2023-03-21:
Thanks for the backup of the watch data. I think I've traced it down to the 'set time' code that's used all over the place. We send the following:

```
(s=>{s&&(s.timezone=TIMEZONE)&&require('Storage').write('setting.json',s);})(require('Storage').readJSON('setting.json',1))
```

The idea is it updates `setting.json` with the correct timezone. The problem would appear to be that if timezone is 0, `s.timezone=TIMEZONE` ends up being 0, which means it's never written to storage!

I'm honestly not sure how this has been an issue for so long and hasn't really been noticed before.

For the moment, please can you just go into Settings in the watch and set the timezone to 0, and then everything should be fine until the timezone gets set to something nonzero and then needs to be set back (by which time I'll have updated the app loader/ide/gadgetbridge to fix this properly)

closed 2023-03-21 by gfwilliams