espruino/BangleApps #2602: Unable to open the menu

tannercollin, 2023-02-24:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

I'm unable to open the menu by clicking the button.

Steps to reproduce: click the button.

Expected behavior: the menu should open.

Instead a "Loading..." box appears in the middle and disappears immediately, or after a few seconds, or after the back light times out. It varies randomly.

I've tried restarting the watch by holding the button down. I've tried doing "Remove all Apps" and "Install default apps" in the App Loader and I still have this issue. One time I was able to get the menu open and did a factory reset, but after the problem remained.

Please me know how I can help debug this issue so I can get my watch working again!

[image]

[image]

[image]

### Installed apps

Only the default ones since I did a "Install default apps" trying to debug this problem.

tannercollin, 2023-02-24:
After sending debug info to log and restarting the watch, it only contains:

```
00000000: 3c 2d 20 42 6c 75 65 74 6f 6f 74 68 0d 0a 3e 0d <- Bluetooth..>.
00000010: 1b 5b 4a 2d 3e 20 42 6c 75 65 74 6f 6f 74 68 0d .[J-> Bluetooth.
00000020: 0a 3c 2d 20 54 65 72 6d 69 6e 61 6c 0d 0a 0d 0a .<- Terminal....
00000030: 48 45 41 4c 54 48 20 45 52 52 3a 20 41 6c 72 65 HEALTH ERR: Alre
00000040: 61 64 79 20 77 72 69 74 74 65 6e 21 0d 0a 3e 0d ady written!..>.
00000050: 1b 5b 4a 2d 3e 20 42 6c 75 65 74 6f 6f 74 68 0d .[J-> Bluetooth.
00000060: 0a                                              .
```

gfwilliams, 2023-02-27:
Thanks - what happens if you connect with the IDE and try and enter the launcher? Does anything get displayed?

Maybe try logging in with the IDE and run `Bangle.factoryReset()` (this will delete everything) and see if it works then?

tannercollin, 2023-02-28:
The problem seems intermittent. I came back to the factory-reset watch after the weekend and it was able to consistently open the launcher right away with no problem.

Then I reuploaded my [modified apps](https://git.tannercollin.com/tanner/bangle) and left the watch to charge all day. When I came back to the watch, the problem was happening again.

I figured it was because of my code so I did a `Bangle.factoryReset()`, but the problem remains. Here's the output of the reset:

```

 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v16 (c) 2021 G.Williams
>Bangle.factoryReset()
Erasing Storage Area...
Erase complete.
Writing initial storage contents...
Write complete.
=undefined
Uncaught SyntaxError: Got [ERASED] expected EOF
 at line 1 col 1
[ERASED]
       ^
in function called from system
>
 ____                 _
|  __|___ ___ ___ _ _|_|___ ___
|  __|_ -| . |  _| | | |   | . |
|____|___|  _|_| |___|_|_|_|___|
         |_| espruino.com
 2v16 (c) 2021 G.Williams
Uncaught Storage Updated!
> 
```

> what happens if you connect with the IDE and try and enter the launcher? Does anything get displayed?

Nothing gets displayed to console when I press the hardware button. Is that what you mean?

Is there a way to open the launcher from the console?

gfwilliams, 2023-02-28:
Wow, that's strange. The `Uncaught SyntaxError: Got [ERASED] expected EOF` after a factory reset is expected, so I don't think there's anything worrying in the output I see there.

If you hard reboot (hold the button for ~10 seconds) does it fix it?

Please could you connect with the IDE, run `reset()` and then `setWatch(print,BTN,{repeat:1,edge:0})`, then tap the button a few times (don't hold it down) and see what happens? You should get a `{ "state": true, ...` message as soon as you press the button, and a `{ "state": false, ` as soon as you release it - if the `{ "state": false` is delayed it would explain it.

On up to date firmwares, if you tap the button on the clock, it should 'fast load', so you won't get the 'loading...' message. The loading message and then clearing (going back to the clock) is what I'd expect if you held the button down for 2 seconds.

... so either it's a hardware issue with a 'sticky' switch, **or** it's something in the software you uploaded that I guess could be resetting the pin state on the button, so that the button no longer 'pulls up' to clear its state.

gfwilliams, 2023-02-28:
> Nothing gets displayed to console when I press the hardware button. Is that what you mean?

Yes - if it were being detected and then an error happened, you should see something hopefully.

> Is there a way to open the launcher from the console?

Yes - `load("launch.app.js")`

gfwilliams, 2023-03-01:
Ok, if you're *sure* it's a full reboot after a factory-reset firmware with `Bangle.factoryReset()` then this could be a hardware issue with a sticky button.

When did you get your Bangle? Could you try pressing the button gently and swirling your finger around on it? Sometimes that can loosen things...

Also is it likely it got wet recently? If water got in next to the button, that could potentially cause this sort of thing (although there is a rubber seal for the button)

tannercollin, 2023-03-01:
> if you're sure it's a full reboot

Does this count as a full reboot? This is what I did:

[image]

> When did you get your Bangle?

I got the watch Jan 04, 2023 from Mouser, who resells Adafruit.

> Also is it likely it got wet recently?

Unlikely, but maybe. I always take it off to shower and do dishes. But leave it on to wash my hands. I consciously avoid getting it wet though.

> Could you try pressing the button gently and swirling your finger around on it? Sometimes that can loosen things...

That fixes it for a couple button presses and then it's back to the same issue.

I think you're right about the button being sticky. The "Loading..." box appears after the same delay as if I had just held the button down instead. Is there a way to fix this?

gfwilliams, 2023-03-02:
Wow, ok - that's pretty new then. We had some issues with the very first batch (~Jan 2022) but since then the buttons have been pretty good. While it's not meant to get super wet, it should be more than fine for stuff like rain and handwashing.

When it happens, can you still feel a 'click' when you press the button in, or does it not click any more?

But if it can be fixed by the way you move the button, it's going to be a hardware fault.

All I can suggest is maybe you could leave it somewhere warm for a day or two and see if that solves the issue (if it was water ingress), but otherwise we'll arrange to get it swapped for a working one.

tannercollin, 2023-03-02:
> When it happens, can you still feel a 'click' when you press the button in, or does it not click any more?

Yup, it clicks exactly like how it always has.

> leave it somewhere warm for a day or two

I'll leave it on my house vent (it's been out on my desk for a few days now) and see if that helps. If it doesn't, is there an email you use for warranty?

gfwilliams, 2023-03-03:
Thanks! And yes, just drop a line to contact@espruino.com if it's not better

bobrippling, 2023-11-15:
Are we happy to close this issue as a broken hardware one?

gfwilliams, 2023-11-16:
Yes, thanks!

closed 2023-11-16 by gfwilliams