espruino/BangleApps #2627: [dtlaunch] Error on console and 1st page doesn display images

dapgo, 2023-03-02:
### Affected hardware version

Bangle 1

### Your firmware version

2v16

### The bug

Scenario: After launching the app, 1st page is empty, moving with btn3, so moving to next app, then the icon for second app is displayed and so on.

Expected: 1st page show 6 icons for apps

Checking on IDE, the app start with following errors:

```
2v16 (c) 2021 G.Williams
Uncaught Error: Expecting first argument to be an object or a String
 at line 14 col 211 in dtlaunch.app.js
...con),x+10,y+10,{scale:1.25});g.setColor(-1).setFontAlign(0,-...
                              ^
in function "draw_icon" called from line 15 col 218 in dtlaunch.app.js
...return i;draw_icon(p,i,selected==i);}
                                     ^
in function "drawPage" called from line 22 col 11 in dtlaunch.app.js
drawPage(0);
```

After clicking btn3, error on console:

```
Uncaught Error: Expecting first argument to be an object or a String
 at line 52 col 64 in dtlaunch.app.js
...con),x+10,y+10,{scale:1.25});
                              ^
in function "draw_icon" called from line 77 col 57 in dtlaunch.app.js
    if (oldselected>=0) draw_icon(page,oldselected,false);
                                                        ^
in function called from system
```
[image]
[image]

dapgo, 2023-03-03:
I think the issue is caused by expeting an image icon, but the default Launcher that has not an icon associated.

```
apps
=[
  {
    name: "Launcher",
    type: "launch",
    icon: undefined, sortorder: -10,
    src: "launch.app.js"
   },
```

bobrippling, 2023-11-15:
Yes, this sounds like a good bug fix to have - happy to take a look at any PRs, and feel free to tag in the author(s) of dtlaunch (@thyttan @Ronin0000 @halemmerich I think)