espruino/BangleApps #2635: [Messages Overlay] Causes BW Clock to stop updating and Quick Launch to stop listening for swipes

thyttan, 2023-03-06:
### Affected hardware version

Bangle 2

### Your firmware version

2v16.206

### The bug

@halemmerich 

Sometimes after dismissing one or more messages (not sure if it only happens with many messages or not) BW Clock stops updating and Quick Launch stops reacting to swipes.

I don't manage to reproduce the bug with Gadgetbridge message debugging.

I wonder if it maybe happens when one/many messages has been sitting on the watch for a while, and I clear them several minutes after they arrived. Investigating now, will be back...

### Installed apps

swscroll (0.02)
notify (0.12)
health (0.22)
sched (0.21)
alarm (0.38)
clock_info (0.02)
dtlaunch (0.22)
widminbat (0.02)
podadrem (0.07)
spotrem (0.07)
messages (0.58)
android (0.23)
lightswitch (0.06)
widanclk (0.02)
backswipe (0.02)
widmsggrid (0.04)
torch (0.11)
calculator (0.07)
agenda (0.13)
smpltmr (0.07)
qcenter (0.04)
sleeplog (0.12)
quicklaunch (0.10)
kbmulti (0.05)
fastload (0.03)
widram (0.03)
widbthide (0.01)
clkinfostopw (0.01)
messageicons (0.05)
widdevst (0.02)
recorder (0.22)
widalarm (0.01)
bwclklite (0.32)
game1024 (0.11)
powermanager (0.08)
boot (0.57)
setting (0.59)
runplus (0.18)
messagesoverlay (0.03)

thyttan, 2023-03-06:
Maybe not related to this issue, but this happened after filling up the RAM with a bunch of debug messages:

[image]

halemmerich, 2023-03-06:
Can you try disconnecting GadgetBridge the next time this happens? I have had my watch locking up sometimes in the last days, usually when something sends lots of messages to the watch. For me it usally its Antennapod updating it's notification every second. My setup is similar to yours, berlinclock, quick launch and fastload. After disconnecting GadgetBridge the watch usually comes back to life in a second or two and processes the last few inputs.

thyttan, 2023-03-08:
>I have had my watch locking up sometimes in the last days, usually when something sends lots of messages to the watch. For me it usally its Antennapod updating it's notification every second. 

This sounds similar to my experience. I think I've noticed the problem mainly when many emails have arrived on the watch at the same time/in rapid succession.  IIRC I was able to clear the messages by touching the top bar and when all messages were cleared the watch didn't respond to input (except for long press of hw button).

>Can you try disconnecting GadgetBridge the next time this happens? 

I may try to do this this weekend, right now I've gone back to using "Message UI".

thyttan, 2023-03-20:
After switching to Messages Overlay again I had some variant of the bug again tonight.

I cleared a message (I think maybe only one) and then Quick Launch stopped responding. I didn't notice if BW Clock stopped updating or not, but I could navigate and use the clock infos. Connection to the Web Ide and executing Bangle["#onswipe"] in the console field seemed to indicate only the swipe handler for the clock infos were left out of my usual three (Quick Launch, Back Swipe and Clock Info).

kamilkrz, 2023-05-26:
@thyttan Firstly try to uninstall lightswitch. This can solve issue with Quick Launch. (there is problem with overlaping handlers for swipe that I try to describe corrctly) #2739
This not exact case but closely related as having lightswitch alone without quick launch still drops some errors in IDE when getting back to clock app (via short press)

Second part (that overlays has impact on redrawing watchface) I believe is still present. 
In my case there is Pebble++ and messagesUI or widbt_notify. When getting back from notification, sometimes parts of watchface are not redrawn until refreshed (clock, clock_info) or not refreshed at all (some part of static background).
On the other hand if clock_info is updated during notify, it will cover its own part of the screen.
I still gather valid info for issue but as this is similar I can leave it here as well.
Oh. And I didn't tried fastload yet.

closed 2023-10-28 by thyttan

bobrippling, 2023-10-30:
I don't use gadgetbridge but I have noticed occasionally, if I've not switched apps in some time, that my watch won't respond to taps, only swipes. On inspecting, the `#ontap` array has about 80 entries in, and I wonder if there's an overlap with whatever's repeatedly registering those listeners and what's going on here.

Hard to reproduce, but I'll see if I can get to the bottom of it

gfwilliams, 2023-10-30:
That's interesting - @bobrippling next time it happens maybe you could print out the contents of the `#ontap` array and post it in a new issue? I think if you just `print` it, you should see the contents of the handlers, and that might give some clue as to what code is adding them but not removing them?