espruino/BangleApps #2648: [compass] Compass goes crazy after charging.

pavelmachek, 2023-03-14:
### Affected hardware version

Bangle 2

### Your firmware version

The one from FOSDEM

### The bug

Bangle.GetCompass().heading works... as long as watch are not charged. After charge, heading is confused, but watch reboot can fix it. I guess the issue is with strong magnets in charging cable confusing the calibration.

gfwilliams, 2023-03-15:
Hi,

Thanks - yes, it's the charge port having magnets as you say, and that throws off the calibration. There's a `.resetCompass()` method which puts it back.

What are you using to get the compass heading though? Normally, what happens is if the compass is off when you charge then it's fine, because it's not taking readings and so not messing up the calibration values.

So really you should only see issues if you're actually running a compass app while charging, which seems like it's not that common an occurrence.

pavelmachek, 2023-05-10:
I was doing some experiments with compass, so it is possible both ran at the same time. Not sure what the right solution is here, maybe just ignore the problem.

gfwilliams, 2023-05-11:
I think on the whole it's rare, but if someone came up with a clock face with a compass on it, suddenly people would start complaining.  Just fixed it in firmware

closed 2023-05-11 by gfwilliams