espruino/BangleApps #2661: fillPoly draws incorrect polygon

wagnerf42, 2023-03-27:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

fillpoly draws a polygon on screen while none is expected in the following example:

(i expect nothing since all y coordinates are negative)

```js
let px = 88;
let py = -253;
let x = 12;
let y = -2071;
let ox = 2.99738204658;
let oy = -0.12530309985;

        g.fillPoly([
          px + ox,
          py + oy,
          x + ox,
          y + oy,
          x - ox,
          y - oy,
          px - ox,
          py - oy,
        ]);

```

i have the bug both on the watch and on the emulator

gfwilliams, 2023-03-27:
Ahh, ok - thanks! I think the issue here is likely that the coordinate `y` is *so* negative that it overflows in the drawing routine (which uses 16 bit x16 fixed point values) and becomes positive again.

I guess we could try and clip values as we convert them to fixed point, but it is this something that's happening to you with real code? It feels like probably if you're drawing something at -2000 then it's a bug in the code?

wagnerf42, 2023-03-27:
hi, 

yeah it's actually real code.
i'm still working on the gps and some segments can be several kilometers long. since the display scale is way smaller
it is possible to reach such coordinates.

note that there is no emergency on my side, i can figure a workaround.

gfwilliams, 2023-03-27:
Ok, thanks! Just put a fix in - but you'll need a cutting edge Espruino firmware to fix it (or 2v18 when it's released)

closed 2023-03-27 by gfwilliams