espruino/BangleApps #2663: [Alarms app / Daylight Savings app] Alarms offset by one hour after Daylight savings time

Aracnoss, 2023-03-27:
### Affected hardware version

Bangle 2

### Your firmware version

2v17

### The bug

### Describe the bug
After daylight savings time, the daylight savings app pushed the clock forward an hour (expected), but all my alarms (set before the change) go off an hour after the time they're set to... I.e. alarm for 5:25pm goes off at 6:25pm.

Can't say if this is a bug because of the DST app or alarms not account for DST. Though I'm prone to go with the former since the alarms app is pretty well tested at this point.

### Expected behaviour
When daylight savings app changes the time forwards, alarms should still go off at the right time.

### Steps to reproduce
1. Wait for DST
2. DST app moves clocks forward.
3. Wait for preset alarm.
4. Alarm goes off an hour after expected.

### Extra
I've just set the time via the main app, and am waiting for next alarm. Will update if that goes off an hour ahead or not.

Aracnoss, 2023-03-28:
### Update
After updating the time via the app hub, my alarms are now going off at the correct time.
So the daylight savings app appears to be causing the bug with how it's adjusting the time.

Aracnoss, 2023-03-29:
Only thing I can add to that, that may help, is that I rebooted the watch at one point during the day thinking that might help and the alarms still went off an hour later, but unfortunately I didn't think to try loading the alarms app to check.

gfwilliams, 2023-03-29:
Thanks, yes - that is helpful. Although it's a little surprising...

So just to be clear, it's this app you have? https://banglejs.com/apps/?id=widdst

Aracnoss, 2023-03-29:
Yeah, that' correct. And I'm using the default alarm app (https://banglejs.com/apps/?id=alarm), I also have the scheduler library installed.

nxdefiant, 2023-03-31:
After switching successfully to DST last weekend I set a timer today for ~1h hour, it alerted me exactly 1 hour after the expected time. I can reproduce this with a new timer set to e.g. 1 minute.

But when pasting the code [sched.boot.js](https://github.com/espruino/BangleApps/blob/master/apps/sched/boot.js) into the IDE everything looks good, so I created a a file atest.boot.js with the following contents
`
console.log(new Date());
`
and uploaded it with the IDE with this result:
>
Date: Fri Mar 31 2023 12:21:11 GMT+0100
>new Date()
=Date: Fri Mar 31 2023 13:21:17 GMT+0200

The first date string is from the boot file, the second from the manual command less then a minute later. Please note the DST offset between both calls.

gfwilliams, 2023-03-31:
I believe just renaming to `widdst.0.boot.js` should do it - I've just pushed a fix to the development app loader. Please can you let me know if that fixes it?

nxdefiant, 2023-03-31:
Looks good, timer for one minute went off after... one minute.

commit a39b773a (2023-03-31): widdst 0.04: Give the boot file the highest priority to ensure it runs before sched (fix #2663)