espruino/BangleApps #2668: [Layout] Layout's lazy mode doesn't clear the screen

rockybulwinkle, 2023-03-30:
### Affected hardware version

Bangle 2

### Your firmware version

2v17

### The bug

When I use the Layout module, with lazy set to true, it does not properly clear modified fields.

Here is example code that shows the behavior, modified from the Layout tutorial:
```var Layout = require("Layout");
var layout = new Layout({
  type: "v",
  c: [{
    type: "txt",
    font: "20%",
    label: "",
    id: "count"
  }, ]
}, {
  lazy: true
});

// timeout used to update every second
var drawTimeout;

// number to show on screen
var count = 9;

// update the screen
function draw() {
  var d = new Date();
  // update time and date
  layout.count.label = count;
  count += 1;
  layout.render();
  // schedule a draw for the next minute
  if (drawTimeout) clearTimeout(drawTimeout);
  drawTimeout = setTimeout(function() {
    drawTimeout = undefined;
    draw();
  }, 1000);
}

// update time and draw
g.clear();
draw();
```

Within 2 updates, when it transitions from display 9->10->11, the bug appears:

[image]

It gets worse over time:

[image]

Am I missing something here about how lazy mode is supposed to be used?

rockybulwinkle, 2023-03-30:
I see some apps are using Layout's update method, but this isn't talked about in the Layout tutorial.

gfwilliams, 2023-03-31:
Ahh - so I think the issue is that layout wants to lay out everything on the screen and figure out how much space to allocate to each thing. In the examples on http://www.espruino.com/Bangle.js+Layout we pre-populate the layout with placeholder strings, which are basically the maximum size we expect things to be.

But then (and we don't do this in the examples) we need to let Layout actually lay everything out - either by rendering with .render, or by just calling .update (which lays out but doesn't render).

Then when you just change the label and render, it's that area which gets cleared.

So for example something like this should work:

```
var Layout = require("Layout");
var layout = new Layout({
  type: "v",
  c: [{
    type: "txt",
    font: "20%",
    label: "----",
    id: "count"
  }, ]
}, {
  lazy: true
});
layout.update();
```

The issue in your code is that the first thing you do is set the label to `9` and then render, so layout assumes the size of the area is all that's needed to render a `9`, and that's all it clears.

I'll update the docs

closed 2023-03-31 by gfwilliams

rockybulwinkle, 2023-03-31:
A font I'm using has variable width. I assume when you say maximum width, that is the maximum pixel width, not maximum character width?

gfwilliams, 2023-03-31:
Yes, it's the maximum space on the screen you think text will take up.

In the docs I'd added at https://github.com/espruino/EspruinoDocs/commit/8d4b022f52c03693c09ce95f302dbfddef80fc13 I mention that you can just use `fillx:1` to make the area stretch across the whole screen, which would fix this without you needing a placeholder