espruino/BangleApps #2674: HRM doesn't turn off

yshui, 2023-04-02:
### Affected hardware version

Bangle 2

### Your firmware version

2v16

### The bug

The heart rate monitor is on the whole time, draining battery quickly.

I tried to `Bangle.setHRMPower(false)` but that doesn't work either. Even after restarting the watch (holding the button), it's still on.

Maybe it's a boot app turning it on, how can I diagnose what's turning it on?

bobrippling, 2023-04-03:
If you use [the IDE](https://www.espruino.com/ide/), what's on `Bangle._PWR` ?

yshui, 2023-04-03:
I suspect maybe I reset the watch after HRM is turned on, which cleared the timers what was going to turn it off?

closed 2023-04-03 by gfwilliams