espruino/BangleApps #2677: Sleeplog App creates hundreds of empty files

Moini, 2023-04-05:
### Affected hardware version

Bangle 2

### Your firmware version

2v17

### The bug

### Issue

The sleep log app created hundreds of empty log files. Not all of them appear in the webinterface for the app, where logs are listed and can be deleted from.

I've just deleted files starting from 2007 (I think) up to 2019, in two week steps per file, one by one, and this took ... eternally. When I check the files in the Files app on the bangle, or in the storage view in the IDE, I can see that there are hundreds more that do not appear in the web interface, and that are empty (Storage app shows `[]`) - I see no reasonable way to delete them. The 'Files' app apparently crashed when trying to remove one of them, and the IDE Storage view reloads after deleting every single file, which takes about 20 seconds or longer for one file.

### What should have happened

* in the web interface:
  * all sleeplog log files should show up, not only those with numbers > 1000
  * multiselect to quickly remove multiple log files at once
  * on mobile:
    * buttons must be larger, it's near impossible to hit them in gadgetbridge bangle.js
    * screen rotation mustn't make the app return to the start screen
 * in the IDE:
   * multiselect for files in storage to delete them
   * maybe consider not doing a full reload?
* in the sleeplog app:
  * only logs for existing dates should be created, not for 10 years back into the past
  * empty logs could be autodeleted...     
  
### Additional questions

How do I delete the sleeplog files 0 to 999 via js from the webIDE? What does the term 'trash files' mean (I apparently have 6500kb of trash files) and can that be emptied? I looked in the Software reference, but it seems it is only possible to list them, not to get rid of them?

### Installed apps

Probably too many:

widbt (0.08), widalarm (0.01), setting (0.59), widlock (0.08), notify (0.12), health (0.22), about (0.14), boot (0.57), locale (0.17), lcars (0.26), sched (0.22), messageicons (0.05), messagegui (0.65), widmessages (0.04), messages (0.58), android (0.23), recorder (0.22), calculator (0.07), info (0.03), widmp (0.08), acmaze (0.03), mylocation (0.09), daisy (0.10), weather (0.25), pastel (0.19), wclock (0.04), configurable_clock (0.03), activityreminder (0.11), compass (0.08), dtlaunch (0.23), ptlaunch (0.15), gbmusic (0.11), powermanager (0.08), hrm (0.11), calendar (0.13), swiperclocklaunch (0.04), run (0.15), barometer (0.04), gpsinfo (0.12), sleeplog (0.12), alarm (0.41), widtbat (0.02), openstmap (0.16), qcenter (0.04), fileman (0.03), gallery (0.03), random (0.02), agenda (0.13), smpltmr (0.07)

Moini, 2023-04-05:
I've just downloaded the backup, to easily delete files. All those sleeplog_[number].log files were not in the 'storage' directory, but where the apps are in the base directory - in case that is relevant.

gfwilliams, 2023-04-05:
Trash files are deleted files. It's nothing to worry about (it's just the way storage works) but you can call `.compact` to remove them.

So the storage files were called `sleeplog_[number].log` ? I believe you could run the command `require("Storage").list(/sleeplog_.*.log/).forEach(f => require("Storage").erase(f))` to get rid of them.

@storm64 is this intentional? Just looking at the README on https://github.com/espruino/BangleApps/tree/master/apps/sleeplog it shows:

> sleeplog_1234.log* | old logfiles

Do these old log files contain info that isn't in the main log, and so need to be on the device? Maybe if the data there is no immediately useful to most people we could have a default-off option on whether they are stored?

storm64, 2023-04-05:
Sry @Moini for your troubles.
It seems you found a bug in sleeplog that was unknown before.

[From @Moini](https://github.com/espruino/BangleApps/issues/2677#issue-1654756321):

> only logs for existing dates should be created, not for 10 years back into the past

This is the intended behavior and for now I can't explain why all the files have been created.

To be able to troubleshoot this bug I would like to ask you some questions:
1. Which version of the sleeplog-app do you use? The latest would be v0.12.
2. Can you tell when all these empty files might have been created?
  a. On the first usage of sleeplog.
  b. Regular on usage.
  c. After an update of the firmware or the app.
  d. After you changed the timezone.
3. Have you experienced a situation when the watch was in an unusual time state, e.g. 01.01.2007?

---

[From @Moini](https://github.com/espruino/BangleApps/issues/2677#issue-1654756321):

> Not all of them appear in the webinterface for the app, where logs are listed and can be deleted from.

@gfwilliams is it possible that by using the following code there is a maximum of files that can be listed or empty files are not listed at all?
`require("Storage").list(/^sleeplog_\\d\\d\\d\\d\\.log$/)`

Moini, 2023-04-05:
@gfwilliams Thank you! So that's what trash files are, and how to remove them (could the keyword 'trash' be added somewhere to the 'compact' function description? I'd have found it that way.). Very good to know! After 'restoring' the modified backup zip, it was gone, too. But if it's not relevant and just overwritten when the space is needed by some app, then I'll not worry about that :)

@storm64

Thank you for getting to this so quickly!

Note that the numbers were **not four**! Those that did not appear in the web interface had only 3 numbers (0 - 999)!

> Which version of the sleeplog-app do you use? The latest would be v0.12.

That's the one I currently have installed, and it is also in my backup from March 24, where those sleeplog_[num].log files are already existing. 

I'm not sure whether the app is installed by default - if so, it was on the watch already when I got it, and when it did not know what time it is yet. Could be that that 'before first connect' time was 2007, I have some very faint memory... Gordon probably knows for sure. Maybe I opened the app before ever connecting via bluetooth? I don't know, but it's possible. I tend to explore everything first :)

**Edit:** the first sleeplog data that I can see in the app is from March 19, so I probably installed the app only on the 18th or 19th. Simply starting the app does not create those files.

Now, it could be that I have used the wrong backup file for restoring yesterday. Or the files were really created again. I will check again tomorrow. I had kept files 0 and 999 last time, in case they were in some way relevant that I did not understand. This time, I have removed them all.

> Can you tell when all these empty files might have been created?

Before March 24, and after March 17, which is when I got the watch. I got it from berrybase.de, in case that helps pinpoint the version number of a potentially installed default sleeplog version.

> a. On the first usage of sleeplog.
- Possibly.

> b. Regular on usage.
~~No. Although I can't say for sure, as I have only removed them all yesterday.~~ 

Oh no - I just checked. And **they're all back**... :-( The first visible file in the web interface is 1000, which seems to correspond to May 1 2008, so I had that wrong with the 2007, it was 2008. The files app shows lots more files, starting at 0 (one digit, not four!)

[image]

Extracted backup in file manager:

[image]

Trying again now. Backing up. Deleting log files from extracted backup. Zipping up. Restoring backup file without those sleeplog files.

[image]

File manager now shows 219 files. And no new files after starting the sleeplog app and closing it. Did I use the wrong (unedited) backup file yesterday...?

>  c. After an update of the firmware or the app.
- Possible.

>     d. After you changed the timezone.
- Possible.

>     Have you experienced a situation when the watch was in an unusual time state, e.g. 01.01.2007?
- I think that could have been the first-time time, when I turned it on after loading. Possibly I also did not set the time right after connecting, but clicked around a little first, installed things and took a look at that cornucopia of things that were available.

Sorry for the chaos - I hadn't expected issues, so I did not take notes ...

Moini, 2023-04-06:
@storm64 : 

Okay, so: after I had uploaded the backup file without any empty sleeplog files except for the main one yesterday, I had checked on my watch and it said it had 219 files. 

Today, it has 547 files again, in the IDE I can see sleeplog files from 0 to 327. So that means they are indeed created freshly and do not stem from the first run where maybe the date was off.

[image]

storm64, 2023-04-08:
This is really strange, but I might have an idea of what could be the cause.
Could you please send me the content of the _sleeplog.log (Storage File)_, as this should trigger the creation of all these files.

If you have concerns about your personal data, the first some lines would be enough. 

Normally it looks like this:
```
2800728,2,1
2800786,4,1
2800787,3,1
2800788,4,1
2800789,2,1
2800794,4,2
2800799,0,02800800,3,2
2800802,4,2
```

Moini, 2023-04-12:
@storm64 Sure, thank you:

time,sleep,consecutive
1680436800,2,1
1680491400,4,2
1680493200,3,2
1680495000,4,2
1680496800,3,2

Moini, 2023-04-29:
@storm64 Would the above confirm the suspicion you had? (and: could I fix it by just removing the app and all associated files, and reinstalling it?)

gfwilliams, 2023-05-18:
Is this fixed now? I know some changes did go in to try and sort out about these files?

Moini, 2023-05-18:
> Is this fixed now? I know some changes did go in to try and sort out about these files?

@gfwilliams I don't know whether the original bug has been fixed. It looked like @storm64 suspected an issue with my sleeplog file, and that might have been the cause.

However, as the huge number of files was slowing down the Bangle, I've opted for deleting all files that have to do with the sleeplog. It's not worth anything anyway, as the heart rate detection is so inaccurate. 

I've reinstalled it, but I'm not wearing the Bangle at night anymore, because the sleep log quality isn't good enough to warrant wearing the watch at all times. 

Although I think heart rate detection has improved somewhat with last week's updates. Maybe it's my circulation.

storm64, 2023-05-28:
Sry for this late response.
My suspicion of the cause inside the logfile was wrong. Your logfile does not seem to be the problem.

The changes I pushed only help to delete the files more easily.

As it doesn't seem to happen to any other user and I can't reproduce the misbehavior on my own hardware, I can't explain what is happening.

Moini, 2023-05-28:
Okay, thanks for checking, @storm64 - then let's close this and hope it won't reoccur.

closed 2023-05-28 by Moini