espruino/BangleApps #2689: messages light - crashes when started up

hughbarney, 2023-04-11:
### Affected hardware version

Bangle 2

### Your firmware version

2v16.93

### The bug

install  Android
test that messages app works using the GadgetBridge debug feature (send a test message from WhatsApp)
no install messages light
send a test message - you will get a blank screen.
long press to exit the messages screen
disconnect from GB and connect to the IDE (so you can see any error messages)
select the launcher and select 'messages light'
you will get a blank screen
In the IDE you will see

```
 2v16.93 (c) 2021 G.Williams
Uncaught ReferenceError: "music" is not defined
 at line 485 col 16 in messages_light.app.js
    if( music!=undefined && EventQueue.length==0 ) return; /...
               ^
in function "updateTimeout" called from line 146 col 17 in messages_light.app.js
  updateTimeout();
                ^
in function "showMessage" called from line 86 col 28 in messages_light.app.js
          showMessage(event);
                           ^
in function "manageEvent" called from line 40 col 34 in messages_light.listener.js
manageEvent(event);
                 ^
in function "callApp" called from line 45 col 22 in messages_light.listener.js
        callApp(event);
                     ^
in function "listener" called from line 328 col 93 in .boot0
require("messages_light.listener.js").listener(type, msg)
                                                        ^
in function called from system
Uncaught ReferenceError: "music" is not defined
 at line 485 col 16 in messages_light.app.js
    if( music!=undefined && EventQueue.length==0 ) return; /...
               ^
in function "updateTimeout" called from line 455 col 17 in messages_light.app.js
updateTimeout();
              ^
in function called from system
> 
```

```

### Installed apps

clkinfosunrise (0.03), widbata (0.04), widpb (0.05), mylocation (0.09), widlock (0.08), lato (0.02), clock_info (0.02), simplestpp (0.03), slopeclockpp (0.09), widalarm (0.01), clkinfostopw (0.01), lcdclock (0.02), setting (0.59), boot (0.57), sched (0.22), alarm (0.41), widgps (0.09), gpstouch (0.02), iconlaunch (0.17), gpsinfo (0.12), messageicons (0.05), messagegui (0.65), widmessages (0.04), messages (0.58), android (0.23), gbdebug (0.01), stopwatch (0.04), messages_light (1.5)

bobrippling, 2023-11-12:
@hughbarney did #2690 / messages light 1.6 fix this for you?

gfwilliams, 2024-05-10:
@hughbarney did https://github.com/espruino/BangleApps/pull/2690 / messages light 1.6 fix this for you?