espruino/BangleApps #2704: [Messages Grid Widget] Turns LCD on unnecessarily

thyttan, 2023-04-22:
### Affected hardware version

Bangle 2

### Your firmware version

2v17.39

### The bug

@rigrig 

I though the LCD of my watch lighted up to often and tracked it down to 'Messages Grid Widget'. It makes sense for the screen to light up when a message is automatically displayed (which it does), but not when only updating the widget (which it also does). This seems to mainly happen if there are unread/not deleted messages on the watch so the widget is showing.

These are some of the situations where the LCD will turn on but I think it shouldn't:
1.  The widget is showing on the watch, I press the power button on my phone, the phones lock screen shows and the watch's LCD is activated.
2. The widget is or isn't showing (makes no difference), auto-open is turned off in settings, an e-mail arrives on the phone and is forwarded to the watch which prompts the LCD turning on.
4. The widget is showing, auto-open music is NOT turned on in settings, the music app Spotify is active and the track ends moving automatically to the next one, the LCD of the watch turns on. (I belive this happens with other media apps which also displays a notification)
5.  The widget is showing, auto-open music is NOT turned on in settings, the music app Spotify is active and I toggle play/pause (either on the phone or via the Spotify desktop client), the LCD of the watch turns on. (I know this happens with other media apps which also displays a notification)

I did some testing of 'Messages Widget' and this doesn't seem to happen with it.

My phone runs Android 13. The Bangle is connected to [a version of Bangle.js Gadgetbridge in this PR](https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/2924) with CompanionDevice Pairing.

### Installed apps

notify (0.12)
health (0.22)
clock_info (0.02)
widminbat (0.02)
podadrem (0.07)
spotrem (0.07)
messages (0.58)
android (0.23)
lightswitch (0.06)
widanclk (0.02)
backswipe (0.02)
torch (0.11)
calculator (0.07)
agenda (0.13)
smpltmr (0.07)
kbmulti (0.05)
widram (0.03)
widbthide (0.01)
messageicons (0.05)
recorder (0.22)
widalarm (0.01)
boot (0.57)
setting (0.59)
runplus (0.18)
sched (0.22)
dtlaunch (0.23)
agpsdata (0.06)
messagesmusic (0.05)
alarm (0.41)
quicklaunch (0.14)
hrmmar (0.02)
taglaunch (0.03)
swscroll (0.03)
messagegui (0.65)
longpressbuzz (0.01)
clkinfostopw (0.02)
sleeplog (0.14)
bwclklite (0.33)
bwclk (0.32)
widmsggrid (0.04)

thyttan, 2023-04-22:
It seems it's the `      Bangle.setLCDPower(1); // turns screen on` on line 73 in the widjet.js code. Maybe some conditions could be checked to decide if the LCD should be activated or not.

rigrig, 2023-04-23:
Thanks for looking into this!   
There's really no need for the widget to turn on the screen, so I made a PR to [remove that line](https://github.com/espruino/BangleApps/pull/2705).

closed 2023-04-24 by gfwilliams