espruino/BangleApps #2712: [sched/alarm] Setting multiple alarms causes only the last to fire

sverona, 2023-04-28:
### Affected hardware version

Bangle 2

### Your firmware version

2v17

### The bug

### Description
When two alarms are set on the same day, eg at 1200 and 1800, only the later alarm fires. On 2v16, *both* alarms fired at 1800. On 2v16 *and* 2v17, nothing happens at 1200.

### Expected behavior
In the above example, one alarm should fire at 1200 and one at 1800.

### To reproduce
1. Set two alarms scheduled for every day at different times.
2. Set the time appropriately (eg, 1159 and 1759)
3. Nothing should happen at 1200. The later alarm should fire at 1800.

### Installed apps

Tested from factory reset.

nxdefiant, 2023-04-28:
Can not reproduce with firmware 2v17 and 2v17.54, alarm 0.41 and sched 0.22.

Tested with timer at 15:00 and 15:05 (CEST).

ticalc-travis, 2026-04-18:
I've just been sticking `console.log` calls for variables in key places of the `sched` code with the Bangle attached to the IDE while exercising the alarm system until the glitch occurs to try to get some clues. I suspect this probably has something to do with specific timing of events.

I'm seeing two possible issues:
1. When alarms use the .js field, somehow the wrong handler gets called sometimes. Apparently this seems more likely to occur when there is more than one alarm triggering at the same time, or in quick succession.
2. When playing around with the IDE attached, switching between apps, running code, etc., it seems that occasionally the timeout for the next alarm that sched.boot.js sets gets “lost” and doesn't fire for some reason? Haven't figured this one out, but maybe it could explain the “missed” alarms described in the first post.

Not sure if these two are related yet. I wonder if I should create a separate issue for the first one. What I have observed, though, is that if 2 occurs and alarms go significantly past due, issue 1 seems more likely (but not guaranteed) to occur upon rebooting or switching apps to reload sched's boot.js handler, causing all those past-due alarms to fire at one time. (One way to simulate this may be to leave the generic alarm message from `sched.js` on screen for a while after it fires, as this seems to prevent further alarms from getting set up and handled until the message is dismissed.)

I'm still working on investigating this as time permits, but looking at the `sched` code and the console output, one red flag sticks out to me. `sched.boot.js` uses some code to build a list of active alarms, earliest first, and sets a timeout to call either the .js handler (if defined) or `sched.js`. But when that timeout fires, `sched.js` then rebuilds its own list of “active” alarms, apparently assuming that the first one in the list will be the same alarm as the first alarm in the active list `boot.js` had generated… yet the code for building the list is not the same. I definitely think there's a possibility that in certain cases the first alarm in the list changes from `boot.js` to `sched.js`, causing `sched.js` to handle a different alarm than intended. So far, this is what my console output seems to suggest. It's like the bug I discovered and recently fixed in tevtimer where I use my own custom alarm handler called using the alarm .js field and realized I can't automatically assume the first alarm in the “active” list is the one I should handle—I need to actually search the list for the alarm my app actually placed.

I think that ideally we would have a more robust way of passing the actual alarm object being handled (or an index or reference of some sort) straight to the alarm handler to save it from having to look it up itself and potentially have issues like this.

gfwilliams, 2026-04-20:
> it seems that occasionally the timeout for the next alarm that sched.boot.js sets gets “lost” and doesn't fire for some reason?

Is this when switching between 'fast load' apps? I guess it's possible that some apps call `clearInterval()` or `clearTimeout()` (without an argument) which would remove all timeouts. Looking quickly at the repo I see several but nothing that would obviously cause issues - unless you have some custom code that does this?

> … yet the code for building the list is not the same. 

Hmm. That's not great. I think we do it this way to avoid loading the entire `sched` library into RAM for the boot code, but the code should really be the same. What is different? That 1 minute check?

I can't remember exactly on the 1 minute thing.  I traced it back to https://github.com/espruino/BangleApps/commit/1dd232ffce46386fe81d6f547bf35a7558e4ca20

I think the idea was (at least at the time) that if you had an alarm that you'd set for 10am and it was 10:30 now, that alarm should go off tomorrow, not now. But I believe that *now* when you set the alarm we set `alarm.last` in `prepareAlarmForSave` if that happens so that check isn't needed? We could just remove it and see if it all still works ok?

> causing sched.js to handle a different alarm than boot.js intended (including possibly one that has its own .js handler that should have been called

Thanks! This is a good catch!

I think from my point of view, it shouldn't really matter if the list of alarms is identical - although in each case you'd hope it was. `sched.js` should take the earliest alarm it finds and handle it. But it feels like it's definitely missing some logic:

* `sched.js` should eval the `.js` field instead of showing an alarm if it exists
* Maybe `sched.js` should allow a second alarm to fire if you were currently displaying the first one? That's a hard one though as what is meant to happen to the first alarm in that case? As minimum I guess when dismissing the first alarm, it should check if an alarm is pending and go straight into displaying that?

> By the way, I discovered that another way to create chaos with the alarm system is to create an alarm that uses the .js attribute but uses it only to do something trivial like console.log and nothing else

Yes, I think that's expected. The readme on it does say:

```
  js : "load('myapp.js')" // OPTIONAL - a JS command to execute when the alarm activates (*instead* of loading 'sched.js')
                          // when this code is run, you're responsible for setting alarm.on=false (or removing the alarm)
```

So I think that's reasonable, although I guess the comment should really say `and rescheduling the alarm`? It's a hard one as I don't think it's a feature that's used that much, so it's not worth putting too much effort into - generally the `js` feature was there to let apps force a load of their own app file for handling, in which case `boot.js` would get called and would reschedule the next alarm.