espruino/BangleApps #2718: [Sleep Log Alarm] Sleep Log Alarm hides Contour Clock

elfehr, 2023-05-03:
### Affected hardware version

Bangle 2

### Your firmware version

2v17

### The bug

### Describe the bug
After installing Sleep Log Alarm and tweaking some of its settings, Contour Clock becomes invisible (only widgets are visible on the clock screen). I suspect it might have been linked to the setting to hide the widget, because the "Zzz" are not hidden when activating it. The clock only reappears when Sleep Log Alarm is disabled or uninstalled, and disappears again after reinstalling / enabling it. I had a bunch of other clocks installed to try them out, among them the default Anton clock, and they were not affected.

### Expected behavior
- Contour Clock never disappears
- Sleep Log Alarm's setting to hide the widget makes it disappear from the clock screen

### Steps to reproduce
1) Have Contour Clock installed.
2) Install Sleep Log Alarm for the first time. At this point all was good: I remember noticing its widget with the time of the next alarm, and nothing was wrong with the clock.
3) Play with Sleep Log Alarm's settings. The clock was still ok after changing the widget color. I suspect the "widget > hide" setting, but I can't be certain because I touched a lot of things before going back to the clock again.
4) Go back to the clock: Contour Clock is invisible. It's not a font problem: in Contour Clock's settings, the text appears as normal in the menu to choose the font.
5) The setting to hide Sleep Log Alarm's widget seems to have no effect.
6) When launching clocks from the menu: Anton clock and Time Traveller's Chronometer are not affected, Contour Clock is. 
7) Reinstall Contour Clock: no effect.
8) Consistently, disabling Sleep Log Alarm from its settings or uninstalling it makes Contour Clock reappears. Enabling / installing it makes Contour Clock disappears again. I can't go back to the initial state when both were working fine after the first installation.

### Installed apps

boot (0.57), health (0.22), setting (0.59), stopwatch (0.04), hrm (0.11), iconlaunch (0.17), widcloselaunch (0.02), scicalc (0.01), sched (0.22), notify (0.12), chargent (0.04), sleeplog (0.14), alarm (0.41), widpedom (0.22), widbthide (0.01), widminbat (0.02), contourclock (0.30), gallifr (0.03), swiperclocklaunch (0.04)

storm64, 2023-05-28:
I just tried to reproduce this issue and couldn't hide the widget, too. While having a look into the code I realized, that I might have wanted to implement this feature, but never added to check for this option on drawing the widget. 😅

Will have a look into this this Sunday.

gfwilliams, 2023-06-05:
So now this is merged, is the issue closed?

elfehr, 2023-06-05:
The new version of Sleep Log Alarm doesn't appear yet on the app loader, so I'm not sure how to check that everything works.
But it sounds like storm64 solved the issue of the hiding the widget (thanks!), but Contour Clock disappearing is a separate problem?

gfwilliams, 2023-06-05:
You should just be able to use https://espruino.github.io/BangleApps/ (the development version of the app loader) to check if you need to...

I just tried this here using apps from that and I can't reproduce it... But if you can, please open this again and maybe let me know exactly what settings you changed so I can reproduce here?

Also, just running and connecting espruino.com/ide when running the clock when it's not working would likely show some error messages that would mean this was easy to track down.

closed 2023-06-05 by gfwilliams