espruino/BangleApps #2739: [lightswitch] problem with drawing widgets when getting back from launcher

kamilkrz, 2023-05-10:
### Affected hardware version

Bangle 2

### Your firmware version

2v17, 2v17.71(Cutting Edge)

### The bug

### Describe the bug
When getting back from launcher (and I think only from launcher) to clock, only lightswitch widget is draw.
No mater if return is done via btn or timeout. Long btn press works fine, getting back from settings works fine.

### Error from IDE:

`Uncaught SyntaxError: Expecting a valid value, got var
 at line 1 col 1 in .bootcde
^
in function "draw" called from line 266 col 12 in lightswitch.wid.js
    w.draw()
           ^
in function called from system`

### Expected behavior
A clear and concise description of what you expected to happen.

### Steps to reproduce
1. Tested with default launcher and dtlauncher
2. Open launcher
3. back to clock with short btn or timeout
4. bug occurs

### Installed apps

- widalarm (0.01)
- widmessages (0.05)
- widbt_notify (0.17)
- chimer (0.02)
- lightswitch (0.06)
- widslimbat (0.01)
- accelgraph (0.01)
- alarm (0.41)
- android (0.23)
- files (0.08)
- authentiwatch (0.07)
- boot (0.57)
- calculator (0.07)
- calendar (0.13)
- clock_info (0.02)
- dtlaunch (0.23)
- health (0.23)
- ha (0.09)
- keytimer (0.02)
- locale (0.17)
- lato (0.02)
- messageicons (0.05)
- messagegui (0.65)
- messages (0.58)
- messages_light (1.6)
- gbmusic (0.11)
- mylocation (0.09)
- notify (0.12)
- owmweather (0.02)
- qcenter (0.04)
- quicklaunch (0.14)
- sched (0.22)
- setting (0.60)
- swscroll (0.03)
- weather (0.25)

storm64, 2023-05-28:
I experienced this bug in development, too. But I thought to have this fixed back then.
Maybe there have been some changes on widget drawing lately.

storm64, 2023-05-29:
Sadly I couldn't reproduce the bug on my hardware, but maybe the PR #2784 might fix this behavior.

kamilkrz, 2023-05-30:
> Sadly I couldn't reproduce the bug on my hardware, but maybe the PR #2784 might fix this behavior.

Side newbie question. For testing purposes how to open your BangleApps directly to lightswitch branch so I can test it beforehand?
Right now Your github.io is pointing to master I think.

storm64, 2023-05-30:
I pushed the changes to master. Now you can test the changes here: [https://storm64.github.io/BangleApps/?id=lightswitch](https://storm64.github.io/BangleApps/?id=lightswitch)

kamilkrz, 2023-05-30:
Yup changes are now visible but no luck. 
When getting back from launcher by short click, only lightswitch widget is present. 
I will provide more debug info later the night as I'm busy now. Thanks for quick reaction

storm64, 2023-05-31:
I just found that @thyttan did a PR #2756  to implement some changes related to a new interpreter feature.
Maybe this could resolve this issue, but I think it is unlikely as it doesn't touch anything related to the drawing process.

bobrippling, 2023-09-15:
I believe this may have been fixed by #3007, @storm64 do you see any problems with the latest version?

bobrippling, 2023-11-15:
Going to close as fixed, but please reopen if you do see the issue after upgrading

closed 2023-11-15 by bobrippling