espruino/BangleApps #2754: [Light Switch Widget / Quick Launcher] Swipe to change brightness triggers quicklaunch as well

thyttan, 2023-05-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v17.106

### The bug

Tapping the Light Switch Widget and then draging changes the brightness like it should, but the swipe handler of quick launcher is also triggered which ideally wouldn't happen.

To fix this I've tried to make use of the new function `E.stopEventPropagation()` introduced in https://github.com/espruino/Espruino/commit/35c8cb9be11ce76ea1829a4a8b40f2ab051bb60b but I can't manage to make Light Switch Widget stop Quick Launcher from acting also, or vice versa. 

I added `E.stopEventPropagation();` inside the `dragListener` function 
in Light Switch Widget [here](https://github.com/espruino/BangleApps/blob/master/apps/lightswitch/widget.js#L144) and inside the swipe handler in Quick Launcher [here](https://github.com/espruino/BangleApps/blob/master/apps/quicklaunch/boot.js#L34). But neither try stopped the other app from executing. 

From the [documentation](https://github.com/espruino/Espruino/blob/35c8cb9be11ce76ea1829a4a8b40f2ab051bb60b/src/jswrap_espruino.c#L2307):
> When using events with `X.on('foo', function() { ... })`
> and then `X.emit('foo')` you might want to stop subsequent
> event handlers from being executed.
> Calling this function doing the execution of events will
> ensure that no subsequent event handlers are executed.
> ```
> var X = {}; // in Espruino all objects are EventEmitters
> X.on('foo', function() { print("A"); })
> X.on('foo', function() { print("B"); E.stopEventPropagation(); })
> X.on('foo', function() { print("C"); })
> X.emit('foo');
> // prints A,B but not C
> ```

First mentioned in #2743:

> Actually no need - I've just implemented option 3 in [35c8cb9be11ce76ea1829a4a8b40f2ab051bb60b](https://github.com/espruino/Espruino/commit/35c8cb9be11ce76ea1829a4a8b40f2ab051bb60b) in the Espruino repo.
> 
> So you just need:
> 
> ```
> E.stopEventPropagation&&E.stopEventPropagation();
> ```
> 
> And that should do it I hope, while being backwards compatible.

_Originally posted by @gfwilliams in https://github.com/espruino/BangleApps/issues/2743#issuecomment-1545357023_

### Installed apps

notify (0.12)
podadrem (0.07)
spotrem (0.07)
android (0.23)
lightswitch (0.06)
widanclk (0.02)
backswipe (0.02)
torch (0.11)
calculator (0.07)
agenda (0.13)
smpltmr (0.07)
kbmulti (0.05)
widram (0.03)
widbthide (0.01)
messageicons (0.05)
boot (0.57)
runplus (0.18)
dtlaunch (0.23)
alarm (0.41)
hrmmar (0.02)
taglaunch (0.03)
swscroll (0.03)
messagegui (0.65)
longpressbuzz (0.01)
bwclklite (0.33)
bwclk (0.32)
widmessages (0.05)
clkinfostopw (0.03)
recorder (0.23)
health (0.23)
quicklaunch (0.15)
polymath (0.01)
sched (0.22)
fastreset (0.01)
setting (0.60)
clock_info (0.03)
messages (0.59)
drained (0.03)
widminbat (0.03)
powermanager (0.10)
pebblepp (0.01)
fastload (0.04)

thyttan, 2023-05-14:
I closed the issue for now since I mixed up swipe and drag events when thinking about this.

closed 2023-05-14 by thyttan

gfwilliams, 2023-05-15:
I guess you need to do something a bit like:

```
dragUsed = false;
Bangle.on('drag', {
  if (light switch widget is doing something) {  
     dragUsed = true;
    setTimeout(function() { dragUsed=false; }, 500);
  }   
});
Bangle.on('swipe', {
  if (dragUsed) 
    E.stopEventPropagation&&E.stopEventPropagation();
  dragUsed = false;
});
```