espruino/BangleApps #2767: [pebblepp] Widgest are not rotated when Lefty flip enabled.

kamilkrz, 2023-05-20:
### Affected hardware version

Bangle 2

### Your firmware version

2v17, 2v17.71(Cutting Edge)

### The bug

The bug
"Slide down" gesture to show hidden widgets shows wrong when clock is set up in lefty LCD orientation

gfwilliams, 2023-05-22:
Hi, thanks - so just to be clear, what actually happens? They respond to a swipe down from the top, but then they slide in from the bottom?

The may be a firmware issue since the overlay is handled in that, and I guess it should probably honour the system rotation setting

gfwilliams, 2023-05-23:
Ok, thanks - I've just put a fix in but you'll have to update your Bangle to the latest 'cutting edge' firmware fot it

closed 2023-05-23 by gfwilliams