espruino/BangleApps #2795: [Feature request] Improve interfacing of alarms between Gadgetbridge and Bangle.js

thyttan, 2023-06-03:
Sync alarms from the scheduler library on the bangle so they show up in the device alarms for the bangle device in Gadgetbridge. And vice versa?

How should creating and deleting alarms via Gadgetbridge and on the bangle impact each other? When connected and when disconnected?

Currently adding an alarm via gadgetbridge and selecting all 7 days of the week sets an alarm on the bangle, but only to be run once - not all days.

This issue is duplicated on BangleApps repo on github and on Gadgetbridge repo on codeberg.

Also reference https://github.com/espruino/BangleApps/pull/2782 introducing setting alarms and timers via the BangleApps App Loader.

thyttan, 2023-06-03:
> How should creating and deleting alarms via Gadgetbridge and on the bangle impact each other? When connected and when disconnected?

An example from Huami devices:

1. GB requests the current alarms state from the watch on connection
2. Changes done in GB while the device is disconnected are lost when connecting to the watch because of (1)
3. When an alarm is changed on the watch, a notification is sent to GB if connected to the watch, so the UI in GB gets updated

Would the bangle support 1 (requesting alarms from watch) and/or 3 (notify GB that changes to the alarms were performed on the watch?

_Originally posted by @joserebelo in https://codeberg.org/Freeyourgadget/Gadgetbridge/issues/3152#issuecomment-928510

gfwilliams, 2023-06-05:
That sounds reasonably doable - it seems like we do something a bit similar for the calendar at the moment, so could use that way of doing things - I think we make the Bangle responsible for sending current timers on connect