espruino/BangleApps #2810: [Developtment SetUp] Integrate pre-commit for Automatic Formatting and Linting

AnotherStranger, 2023-06-11:
**Issue Description:**
I would like to propose the integration of [pre-commit](https://pre-commit.com/) into the project as a tool to automate the process of formatting and linting files. This would help ensure consistent code quality and improve the overall development experience for contributors.

**Overview:**
pre-commit is a versatile and widely used tool that allows developers to define and manage hooks for performing various tasks, such as code formatting, linting, and more. By integrating pre-commit into our project, we can automate these tasks and enforce code quality standards, both during commits and when reviewing pull requests.

**Proposed Changes:**
1. Add pre-commit as a development dependency to our project.
2. Provide a pre-commit configuration file, `pre-commit-config.yaml`, that defines the hooks and tools to be used.
3. Update the project's documentation to include instructions on how to set up and activate pre-commit.

**Benefits:**
1. **Automated Formatting and Linting:** pre-commit will enable automatic formatting and linting of files on each commit. This will help maintain a consistent code style across the project and reduce the burden of manual formatting.
2. **Improved Code Quality:** By enforcing correct linting and formatting on pull requests, we can ensure that all changes submitted for review meet the project's quality standards. This will facilitate easier code reviews and reduce the likelihood of introducing issues related to code style.
3. **Flexibility and Manual Activation:** Developers will have the option to activate pre-commit manually, ensuring that it doesn't impose any changes without consent. They can choose to enable or disable pre-commit based on their preferences and needs.

**Considerations:**
It's important to note that while pre-commit offers numerous hooks for various tasks, I propose focusing on code formatting and linting initially. Additional hooks and tools can be explored and added in the future, based on the needs and requirements of the project.

I welcome feedback and discussion from the community and project maintainers regarding this proposal. Thank you for considering my suggestion.

bobrippling, 2023-06-12:
I'd come down on not being a fan too - while I like being able to write apps in typescript and all the clunking/npm wrangling that comes with it, I do enjoy that I can code something with almost no barrier to entry at the same time.

I do run formatters on some of my code but keep it just to an individual app, and outside of the repo if that helps?

AnotherStranger, 2023-06-17:
Hello @gfwilliams and @bobrippling !

At first, I'd like to apologize for not coming back to you earlier!
I think I may have introduced a misunderstanding here. By default, pre-commit does not enforce any coding style but is more like a convenient tool for people to use. I just wanted to mention that one _can_ use it to enforce coding or formatting rules. However, I see your point and have to agree with you: It's more important to keep contributing as easy as possible.

My proposal was more meant like this:
1. I could create a pre-commit config file which does nothing on its own
2. If someone wants to use these hooks he or she can install pre-commit and activate them using `pre-commit install`.

If you like to see what I have in mind I will gladly create a branch for demonstration purposes.

bobrippling, 2023-11-15:
I suppose in that case it's more down to an individual dev's setup - we could see if other people would find that useful and decide whether to then bring it into the repo though

closed 2024-01-16 by AnotherStranger