espruino/BangleApps #2816: [E.showMenu] Title of selected item in options array appears in subsequent list

philipandresen, 2023-06-12:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

### Description
When using the array version of menus along with E.showMenu (as described in the docs), while navigating from one menu to another, the selected item on the first menu appears in the second menu.

### Expected Behaviour
The second menu renders normally and the first menu can no longer be seen.

### Reproduction steps
1. Open web IDE, write the following code. Note that only the first menu needs to be the "array" type to reproduce.
```
let menu = [
  { title:"A"},
  { title:"B"},
  { title:"C", onchange:function() { E.showMenu(submenu); } },
];
menu[""] = { title:"Hello" };

let submenu = {
  "" : { title : "World", back : function() { E.showMenu(menu); } },
  "One" : undefined,
  "Two" : undefined,
  "Three" : undefined,
  "Four" : undefined
};

E.showMenu(menu);
```
2. When the code loads, select item "C".
3. Note that the second menu renders normally but with "C" superimposed (this follows scrolling motions too).
4. Note that the menu item that lingers from one to the other is the one you pressed to navigate.

The lingering button cannot be interacted with, but it can be cleared when an element underneath it is pressed such that it refreshes its render.

### Root cause hypothesis:
onchange invokes a refresh of the pressed menu item to allow it to display the "new" value. This refresh might not be synchronous with the other actions being taken, so when you navigate away from the menu, the refresh happens after navigation and causes the button to render over the screen.

I haven't looked into the source code yet but will update if I can identify a specific line.

### Installed apps

Default apps only plus one custom keyboard library [kbmatry]

nxdefiant, 2023-06-12:
There is a known workaround to use setTimeout to have the onchange callback called after the draw and not nice versa.

`{ title:"C", onchange:function() { setTimeout(() => E.showMenu(submenu)) } },`

philipandresen, 2023-06-12:
> There is a known workaround to use setTimeout to have the onchange callback called after the draw and not nice versa.
> 
> `{ title:"C", onchange:function() { setTimeout(() => E.showMenu(submenu)) } },`

That's good to know. Is this issue a duplicate by chance? I did look around but didn't see anything open.

stweedo, 2023-06-12:
I wasn't aware of this bug, but I did run into it in one of my menus and thought I was doing something wrong. Using a setTimeout fixed it.
```
setTimeout(() => {
  E.showMenu(createMainMenu());
},(0));
```
Thank you :)

gfwilliams, 2023-06-13:
Thanks - in the old menu style (`{ "text": function}`) this shouldn't be an issue - as @philipandresen noted it seems this only happens for menus where you have an `onchange` handler.

I think the nicest option is actually to make the menu detect if it's still 'active' (eg if `setUI` was called during the `onChange` handler) and only redraw if so... so that should fix any other issues in code where `setTimeout` had to be added as well.

I've just put a fix in for this, so I think we're good now

closed 2023-06-13 by gfwilliams