espruino/BangleApps #2822: [Authenticator] Does not pick up the QR code

danielo515, 2023-06-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

The web app does not pick up the QR code of a google authenticator export.
I will love to be able to take all the accounts that I have there to my two bangle.js watches.

### Installed apps

It doesn't matter IMO, it is a problem with the configuration web app

bobrippling, 2023-06-14:
Google auth uses some non-standard extensions that a lot of open source tools don't support (such as [Authy](https://authy.com/)), and I suspect it's the same here.

passivedragon, 2023-07-16:
A workaround is to use a plain QR reader, extract the secret from the text and just use that to setup directly

danielo515, 2023-07-17:
Yes, I did that on the past for other apps, but it s very inconvenient