espruino/BangleApps #2832: [compass] Can't exit from the compass app

UbioZur, 2023-06-21:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

### Describe the bug
Once I start the compass app, I can't seem to find a way to exit from it.

### Expected behaviour
Some kind of swipe on the screen would allow to quit the app.(as the button is used for reset)

### Expected User Experience
Every other apps are quit by pressing the button once. So I would expect to quit the compass app by pressing the button once... 
- single press button: quit the app
- long press button: recalibrate?
- long press on the touch screen: recalibrate?

### Installed apps

- Compass
- BW Clock
- Desktop Launcher
- Health
- Barometer

nxdefiant, 2023-06-21:
try a longer long press ;)

gfwilliams, 2023-06-21:
Yes - I think just holding the button for ~1 second (until the `Loading...` message appears) should have the desired effect. This is how you can be sure of exiting all apps.

As I recall a short press of the button is used to reset calibration.

UbioZur, 2023-06-21:
Thanks for the information, at least now I know I can long press the button to quit any app.

The long press (1-2s) of the button does indeed quit the app, Sadly it also seem to reset the calibration, meaning that I have to re calibrate every time I open the compass!

closed 2023-06-21 by UbioZur

gfwilliams, 2023-06-22:
Ahh - thanks for letting me know! I just fixed that in the development app loader