espruino/BangleApps #2836: [android] Emoji/text messages get corrupted

halemmerich, 2023-06-23:
### Affected hardware version

Bangle 2

### Your firmware version

2v18.69

### The bug

[image]

To reproduce send a GB debug message containing text and emoji. Some combination work fine, others don't:

Working:
a
😀😀
Garbled:
😀
😀😀😀
😀😀a
a😀😀
a😀a😀

### Installed apps

agpsdata (0.06), alarm (0.41), bthrm (0.17), choozi (0.04), drained (0.03), drinkcounter (0.25), health (0.23), iconlaunch (0.17), imageclock (0.13), info (0.03), locale (0.17), messageicons (0.05), mylocation (0.09), powermanager (0.10), qcenter (0.04), qmsched (0.09), quicklaunch (0.14), waypoints (0.01), widalarm (0.01), widclk (0.07), widcloselaunch (0.02), widlock (0.08), widminbat (0.03), widminbt (0.01), owmweather (0.03), keytimer (0.03), qrcode (0.06), loadingscreen (0.01), boot (0.58), setting (0.61), runplus (0.21), notify (0.13), messages (0.60), kbmulti (0.06), sched (0.22), sensortools (0.05), gpstrek (0.10), metronome (0.07), wrkmem (1.01), gipy (0.18), fastload (0.03), android (0.28), messagegui (0.72), recorder (0.28)

gfwilliams, 2023-06-26:
Ok, thanks - I'll take a look.

It's to do with the Unicode support that got added to Espruino - some binary bits of image can fall into the unicode range, and while it should be able to handle that ok (converting everything to unicode) it looks like something's going wrong.

gfwilliams, 2023-06-26:
Just for reproduction, this can be pasted into the IDE or even run on Linux:

```
g.drawString(eval(E.toString([0x22,0x5c,0x30,0x5c,0x78,0x31,0x37,0x5c,0x78,0x31,0x35,0x83,
0x5c,0x31,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,
0x24,0x93,0xa6,0x9a,0x69,0x5c,0x74,0x24,0x92,0x49,0x26,0x69,0xa6,0x9a,0x69,0x89,0x24,0x92,0x49,0x9a,
0x69,0xa6,0x9a,0x69,0xa1,0x24,0x92,0x66,0x9b,0x6d,0xb6,0xda,0x69,0x89,0x24,0x93,0xa6,0x9a,0x69,0xa6,
0x9a,0x69,0x5c,0x74,0x24,0xe9,0xb6,0x5c,0x33,0x6d,0x80,0x9a,0x68,0x49,0x26,0x69,0x80,0x5c,0x78,0x31,
0x61,0x60,0x5c,0x30,0x9a,0x68,0x49,0x3a,0x6d,0x80,0xdb,0x60,0x5c,0x36,0xda,0x62,0x49,0x9a,0x69,0x86,
0x9a,0x61,0xa6,0x9a,0x5c,0x78,0x31,0x32,0x4e,0x9b,0x6d,0xb6,0xdb,0x6d,0xb6,0x98,0x92,0x66,0x9a,0x69,
0xa6,0x9a,0x69,0xa6,0x5c,0x34,0x93,0xa6,0xd8,0x69,0xb6,0x5c,0x78,0x31,0x33,0x69,0xa1,0x24,0x99,0xa6,
0x98,0x5c,0x30,0x5c,0x30,0x5c,0x78,0x31,0x61,0x69,0x5c,0x74,0x24,0x99,0xa6,0xd8,0x41,0xa6,0xda,0x68,
0x49,0x24,0xc9,0xa6,0x9a,0x69,0xa6,0x98,0x5c,0x78,0x31,0x32,0x49,0x24,0xe9,0xa6,0x9a,0x69,0xa6,0x84,
0x92,0x49,0x24,0x99,0xa6,0x9a,0x69,0x5c,0x31,0x24,0x92,0x49,0x24,0x93,0xa6,0x9a,0x5c,0x78,0x31,0x32,
0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x5c,0x31,0x22])),0,7)
```

But this works:

```
g.drawString(eval(E.toString([0x22,0x5c,0x30,0x5c,0x78,0x31,0x37,0x5c,0x78,0x31,0x35,0x83,
0x5c,0x31,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,
0x24,0x93,0xa6,0x9a,0x69,0x5c,0x74,0x24,0x92,0x49,0x26,0x69,0xa6,0x9a,0x69,0x89,0x24,0x92,0x49,0x9a,
0x69,0xa6,0x9a,0x69,0xa1,0x24,0x92,0x66,0x9b,0x6d,0xb6,0xaa,0x69,0x89,0x24,0x93,0xa6,0x9a,0x69,0xa6,
0x9a,0x69,0x5c,0x74,0x24,0xa9,0xb6,0x5c,0x33,0x6d,0x80,0x9a,0x68,0x49,0x26,0x69,0x80,0x5c,0x78,0x31,
0x61,0x60,0x5c,0x30,0x9a,0x68,0x49,0x3a,0x6d,0x80,0xab,0x60,0x5c,0x36,0xaa,0x62,0x49,0x9a,0x69,0x86,
0x9a,0x61,0xa6,0x9a,0x5c,0x78,0x31,0x32,0x4e,0x9b,0x6d,0xb6,0xab,0x6d,0xb6,0x98,0x92,0x66,0x9a,0x69,
0xa6,0x9a,0x69,0xa6,0x5c,0x34,0x93,0xa6,0xa8,0x69,0xb6,0x5c,0x78,0x31,0x33,0x69,0xa1,0x24,0x99,0xa6,
0x98,0x5c,0x30,0x5c,0x30,0x5c,0x78,0x31,0x61,0x69,0x5c,0x74,0x24,0x99,0xa6,0xa8,0x41,0xa6,0xaa,0x68,
0x49,0x24,0x59,0x56,0x9a,0x69,0xa6,0x98,0x5c,0x78,0x31,0x32,0x49,0x24,0xa9,0xa6,0x9a,0x69,0xa6,0x84,
0x92,0x49,0x24,0x99,0xa6,0x9a,0x69,0x5c,0x31,0x24,0x92,0x49,0x24,0x93,0xa6,0x9a,0x5c,0x78,0x31,0x32,
0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x92,0x49,0x24,0x5c,0x31,0x22])),0,7)
```

In it, I replaced any bytes in the unicode range.

So the issue is that unfortunately the image itself happens to contain not just chars in the unicode range (which would be fine) but actual valid unicode sequences, which encode what should have been multiple characters into one unicode char and that throws everything off.

So I've had to change Gadgetbridge. If you're signed up for testing releases on https://play.google.com/apps/testing/com.espruino.gadgetbridge.banglejs it should all work ok now

halemmerich, 2023-06-26:
Great, thanks. I'm not using the Playstore build, but I will build your master and try this out this evening.

halemmerich, 2023-06-26:
Seems to be working fine now :+1: