espruino/BangleApps #2854: gpstrek: Can not parse gpx

nxdefiant, 2023-07-01:
The following test.gpx created with gpsprune can not be parsed from the gpstrek interface:
[testgpx.zip](https://github.com/espruino/BangleApps/files/11925834/testgpx.zip)

After clicking on "Upload" all I get is

> File could not be converted correctly.
error

Notify @halemmerich

When I replace "GPX/1/0" with "GPX/1/1" in the gpx it seems to work.

halemmerich, 2023-07-01:
The XSL used for conversion references the GPX 1.1 version. Maybe for different versions of GPX files multiple XSL are needed. There seem to be the relevant version 1.0, 1.01 and 1.1. I think the `trkpt` nodes are universally used in all version, so maybe ignoring the namespace in the XSL would be enough?

nxdefiant, 2023-07-01:
The Quick & Dirty solution would be to just search and replace the gpx file.

gfwilliams, 2023-07-14:
sounds good to me!

PR #2875 (2023-07-17): Gpstrek - Handle GPX 1.0 in interface.html
No version bump since no code changes for the Bangle side of things. Fixes #2854
closed 2023-07-17 by gfwilliams