espruino/BangleApps #2860: messages+messagesgui app/library

d3nd3, 2023-07-03:
https://github.com/espruino/BangleApps/blob/55e52be7ddec9830b8b1a364e464958b2b7fa3f6/apps/messages/README.md?plain=1#L38-L51

This statement seems in contradiction with how its implemented.  No event is emitted and:
`Bangle.load((msg && msg.new && msg.id!=="music") ? "messagegui.new.js" : "messagegui.app.js");` is ran.

So I assume one could write their own gui by installing an app on top of messagesgui that uses the same file name `messagegui.app.js` in their app? Or is the event emitting work in progress?

gfwilliams, 2023-07-03:
Yes, that doesn't seem quite right. I believe you can do `require("messages").openGUI()` and that's fine.

But `Bangle.load(...."messagegui.app.js");` is called from messagegui's lib.js listener fn, which is called from boot.js with `Bangle.on("message", (type, msg) => require("messagegui").listener(type, msg));`

So actually the only difference is it should say `Bangle.on("message", (type, msg)` and not `Bangle.on("messageGUI", message=>{ ` ?

d3nd3, 2023-07-03:
https://github.com/espruino/BangleApps/blob/55e52be7ddec9830b8b1a364e464958b2b7fa3f6/apps/messages/lib.js#L109-L120C3

This comment says `emit` but no emit eg.

> But Bangle.load(...."messagegui.app.js"); is called from messagegui's lib.js listener fn
Ah, right I didn't see this pathway.  Do the pathways sit well with each other, hm.

d3nd3, 2023-07-03:
The only thing I don't understand is where is the `openGUI` event emitted from?

gfwilliams, 2023-07-03:
There is no openGUI event? It's just the method... Where did it say there was a openGUI event?

d3nd3, 2023-07-03:
Here:
https://github.com/espruino/BangleApps/blob/55e52be7ddec9830b8b1a364e464958b2b7fa3f6/apps/messages/lib.js#L110

Thats the part that had me confused, hm.

gfwilliams, 2023-07-03:
Ok - yes, it looks like when everything was split into apps the docs weren't updated. I've updated the docs.

Basically if writing a gui you provide your own `messagegui` lib which has a `open` method that gets called

commit cb2760a3 (2023-07-03): docs - fix #2860
closed 2023-07-03 by gfwilliams