espruino/BangleApps #2869: [swiperclocklaunch] Prevents launcher from starting any app

buergi, 2023-07-10:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

After installing swiperclocklaunch opening anything from the launcher fails with this error

```
Uncaught Error: Cannot read property 'startsWith' of undefined
   at line 23 col 228 in .boot0
  ...typeof mode)mode=mode.mode;if(mode.startsWith("clock")){Bangle.swip...
                                       ^
  in function "setUI" called from line 1 col 7377 in messagelist.app.js
  ...ns);Bangle.setUI(options,cb);
                                ^
  in function "setUI" called from line 1 col 14007 in messagelist.app.js
  ...gle.appRect;layout=undefined;setUI();g.reset().clearRect(Bangle.appR...
                                        ^
  in function "clearStuff" called from line 1 col 14084 in messagelist.app.js
  clearStuff();if(active&&screen!==active)back=active;if(screen==="me...
                    ^
  in function "setActive" called from line 75 col 730 in messagelist.app.js
  ...-1;setActive("messages",num);if(!MESSAGES.length){return E.showProm...
                                       ^
  in function "showMessage" called from line 105 col 1226 in messagelist.app.js
  ...ewIdx>=0){showMessage(newIdx);let unreadTimeoutSecs=settings().u...
                                    ^
```

I solved it by swapping lines 11 and 12 in `swiperclocklaunch/boot.js` https://github.com/espruino/BangleApps/blob/edfb9000e8624623106f24f49d2ac16aeeaa19db/apps/swiperclocklaunch/boot.js#L11-L12

### Installed apps

widChargingStatus (0.02), widminbt (0.01), qmsched (0.09), locale (0.17), teatimer (0.03), widstep (0.01), calclock (0.06), hrm (0.11), widalarm (0.01), stopwatch (0.04), messagelist (0.02), clkinfocal (0.01), clkinfosunrise (0.03), qcenter (0.04), messageicons (0.05), agenda (0.13), weatherClock (0.06), weather (0.25), sched (0.22), alarm (0.41), game1024 (0.11), balltastic (0.03), widmessages (0.05), health (0.23), widminbat (0.03), iconlaunch (0.17), notify (0.13), boot (0.58), runplus (0.21), messages (0.60), setting (0.61), kbmatry (1.07), guitar (0.02), clock_info (0.06), android (0.29), spotrem (0.09), bwclk (0.33), mylocation (0.10), widmp (0.08), agpsdata (0.06), gpsinfo (0.12), swiperclocklaunch (0.05)

bobrippling, 2023-07-10:
The question here is whether the bug's in `swiperclocklaunch`, or `messagelist`.

`messagelist` will [call `Bangle.setUI` with an options object](https://github.com/espruino/BangleApps/blob/edfb9000e8624623106f24f49d2ac16aeeaa19db/apps/messagelist/app.js#L89-L89) that doesn't contain `mode` - it looks like [omitting `mode` is intended API](https://www.espruino.com/ReferenceBANGLEJS2#l_Bangle_setUI) so I reckon the fix wants to be in `swiperclocklaunch` - I'll sort a PR

gfwilliams, 2023-07-12:
Thanks! Yes, it seems like `swiperclocklaunch` needed fixing - it should be able to deal with `Bangle.setUI("mode")` or `Bangle.setUI({mode:"mode", ...})`

PR #2871 (2023-07-12): swiperclocklaunch: check `mode` after pulling from `options`
Fixes #2869
closed 2023-07-12 by gfwilliams