espruino/BangleApps #2878: [Messages Overlay] Displayed message keeps screen awake

thyttan, 2023-07-14:
### Affected hardware version

Bangle 2

### Your firmware version

2v18.92

### The bug

### The bug:
When a message has arrived and is displayed on the watch it will keep the screen backlight on. The backlight will turn off and back on instantly with some interval.

Video demonstration:
https://github.com/espruino/BangleApps/assets/97237430/026f13a7-d5d2-402b-88cd-fcdb3ff348a2

### Expected behaviour:
The message arrives and turn the screen on. The backlight times out and stays turned off.

### Reproduce:
1. Install Messages Overlay and it's dependencies.
2. Send a test message from Gadgetbridge debug page.
3. Observe the described behaviour.

### Furter notes:
I uninstalled Fastload Utils and tested the bug prior to this report.

### Installed apps

sched (0.22)
messageicons (0.05)
widmsggrid (0.05)
health (0.23)
widminbat (0.03)
widanclk (0.02)
backswipe (0.02)
torch (0.11)
calculator (0.07)
widram (0.03)
widbthide (0.01)
clkinfostopw (0.03)
dtlaunch (0.23)
quicklaunch (0.15)
alarm (0.41)
powermanager (0.10)
mysetup (0.01)
fastreset (0.01)
bwclklite (0.34)
smpltmr (0.08)
hrm (0.11)
lightswitch (0.07)
setting (0.61)
boot (0.58)
messages (0.60)
dragboard (0.07)
runplus (0.21)
agenda (0.14)
clock_info (0.06)
spotrem (0.09)
recorder (0.29)
messagegui (0.73)
podadrem (0.09)
swscroll (0.05)
gipy (0.20)
android (0.29)
messagesoverlay (0.04)

halemmerich, 2023-07-15:
Interesting, had not seen this before but I can reproduce. It seems to be caused by the LCD being switched on every time a lock event happens. So far so easy, but why didn't this happen before? That part of the code wasn't changed in the last PR. I believe I have a fix, will test a bit and do a PR.

halemmerich, 2023-07-15:
Found the answer, this bug was in 0.03 as well. That will have used quite some battery...

PR #2880 (2023-07-20): messagesoverlay - Fix overlay keeping the LCD on
Fixes #2878 by not switching on the backlight for every lock event.
closed 2023-07-20 by gfwilliams