espruino/BangleApps #2883: [homeassistant] Visually blank, nothing is shown, but still works

passivedragon, 2023-07-16:
### Affected hardware version

Bangle 2

### Your firmware version

2v17, 2v18

### The bug

### Behavior
Upon opening the app on the device, the screen shows nothing, blank, just background. With correct setup however, tapping the screen still sends the first intent. Swiping or any other action appear to do nothing at all, upon tapping it always sends the first intent that was setup in the JSON config.

### Expected behavior
Visuals like the attached screenshot to the apps readme as well as functional switching between intents.

### Reproduction
1. stetup app
2. start app

gfwilliams, 2023-07-20:
@peerdavid  any thoughts on this? it seems like it's not intentional?

peerdavid, 2023-07-21:
I wonder how the correct setup looks like @passivedragon -- sounds a bit like if the json is not correct. Could you share it?

passivedragon, 2023-07-30:
> I wonder how the setup looks like @passivedragon -- sounds a bit like if the json is not correct. Could you share it?

```json
[
  {"display": "Open", "trigger": "OPEN_DOOR", "icon":"door"},
  {"display": "Office", "trigger": "TOGGLE_LIGHT", "icon":"light"},
  {"display": "Living Room", "trigger": "OVEN", "icon":"fire"}
]
```
I believe I haven't edited the default values whatsoever.
Let me know about any other information you'd like me to pull out.

passivedragon, 2023-07-30:
Very odd... My apologies. After reuploading this json without any edits, it now functions.
Interesting observation:
before uploading this json, pressing the blank screen that was present sent an intent labeled `DOOR`
Seems like that might be a default value of some sort, since it doesn't appear  in  the json in that way, other than perhaps the icon textcapitlaized, which i doubt is happening.

So this is at least to some degree user error, but perhaps the default is a bit misleading? It seems odd to me that installing the app resulted in that odd state, despite me never going in to mess with the JSON.

For my purposes my issue is resolved with this, thank you.

peerdavid, 2023-07-31:
Ok so do you think its fine to close this issue?

bobrippling, 2023-11-15:
Let's close for now and re-open if it keeps occurring easily

closed 2023-11-15 by bobrippling