espruino/BangleApps #2892: [Drained] BT connection not possible after Drained shifts into power saving mode

lightnin, 2023-07-21:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

It appears as though my watch will not connect to bluetooth on the laptop of phone (via gadgetbridge app) after the Drained app has shifted into low power mode. 

Workaround: Restore power settings with drained app.

bobrippling, 2023-07-22:
Thanks for the report - that's all [intentional behaviour](https://github.com/espruino/BangleApps/blob/0eea248390f5245c8d5bceb3eb2e976fab193a45/apps/drained/app.ts#L25-L30) to reduce battery usage as much as possible when `drained` is active. Like you say, disabling drained via the usual button method brings back BLE connectivity.

lightnin, 2023-07-22:
Is there a way to make it more clear to the user that this is happening (or will happen)? For example - have the bluetooth widget appear "disabled" in some way, perhaps with a cross through it or some such clear indicator?  Otherwise users may forget that this is the normal behavior for "Drained" and assume there is some other cause for the bluetooth connection problem, as I did. Also, this problem seemed to persist even after I charged the watch fully again. I was on the point of thinking my watch's bluetooth chip had failed....
The other question I would ask is: Is the energy saved worth the potential confusion this may cause?

bobrippling, 2023-07-23:
So `drained` doesn't load any widgets - were you seeing a bluetooth widget while it was running? If so that'll be a bug.

For bluetooth not enabling when your watch was charged, is it that `drained` was running all the way through? Do you have the restore threshold set very high? Or if it's easier, could you paste in your `drained.setting.json`? (this can be read via [the IDE](https://www.espruino.com/ide/#)).

lightnin, 2023-07-24:
No, Drained doesn't display widgets. But when entering the settings menu a BT widget is shown (perhaps it's not a widget, but it does give a kind of BT status).
 
Once I figured out that Drained was the cause of my problems, I deleted Drained and so don't have the settings json. My restore threshold wasn't high - and I did put it through at least one full recharge and the BT functionality did not begin working again by itself. Also note that after a recharge Drained would not restore the screen brightness, even though the settings indicator stated that it was set to 1.0: I had to set it to a lower value, and then back to the high value to get the screen brightness to actually go back to normal. This and the BT connection problems makes me think that there is a bug in the code that restores normal operation after a recharge.

Really, I was at my wits end troubleshooting the watch, and was more or less convinced the hardware BT functionality was dead or failing intermittently. I suppose you could say RTFM. But even if the user did read the manual, one shouldn't expect them to remember this kind of thing weeks after installing the app without some sort of prompt / indicator to help them understand that the BT radio has been powered off by Drained in order to help them in the process of troubleshooting.  

I'm a bit concerned that because this app touches very low level operations in the watch which are normally only accessible via firmware settings, it could result in unnecessary support calls for @gfwilliams  / or possibly even returns. Note that at one point I did a firmware reset, restored the apps I had favorited (including drained), and then still couldn't connect afterwards. I should have realized it was Drained at that point, but didn't because I don't normally think of apps as having the necessary privileges to break BT connectability - something which I would consider to be an "OS" level function.

My recommendation would be to remove the disabling of BT functionality from the Drained app - and that @gfwilliams consider preventing apps from having that privilege. If you don't agree, I propose that the "Make connectable" option in the BT settings menu do something to override any state set by an app like Drained such that the BT subsystem will work.

I can see good reasons for the functionality in Drained - and applaud you contribution. But I do think there are big UX risks here that need to be taken into account.

bobrippling, 2023-07-25:
Interesting, did you have fastload installed perhaps? In which case I can see bluetooth not restarting between app launches, so if Drained had run, bluetooth would be out until the next "full" reload.

No worries about the `settings.json`, I suspect it might've been something like the above - yes like you say, sounds like there's an issue with restoring normal operation in some situation, which I think I know how to fix.
If this does happen in future, there's [this method](https://www.espruino.com/Bangle.js2#resetting-without-loading-any-code) to get connectivity if something's misbehaving at boot time :)

"Make connectable" does wake the NRF (bluetooth) chip back up, so that's intriguing if you did give that a whirl too.

lightnin, 2023-07-26:
It's weird - I got into the no BT connect state again yesterday, which made me less sure of my conclusions. Tried Factory reset, no dice (via phone or laptop). Started trying a lot of things semi-systematically. Finally, it started connecting after I did a Utils -> Rewrite settings. 

One strange thing to note: After the factory reset, I noted that in the bluetooth menu, the "Whitelist" state was set to on. That's the state it is in now, and it is connecting, though since there have been several factory resets since I last played with the whitelist, it's hard to believe it has any data in it. 

Is it possible for Whitelist to get into a state where it's reverse toggling, so the menu item doesn't reflect the true state? Or is my understanding that Whitelist -> on should mean that the watch won't connect to anything not on the whitelist mistaken?

Note that when I choose "add device" while connected to my phone, It says "connect device to add to whitelist" - and it is connected, but no menu entry with the bt address appears. When I connect via my laptop, a menu entry with my laptop's bluetooth address appears, which I then have the option to remove. 
That's weird - I might have set this up a long long time ago, but I wouldn't expect whitelist data to persist after a factory reset (and several firmware upgrades). And even though it says it is on, it will connect to my phone without any entry showing up.

thyttan, 2023-07-26:
>Edit: Sorry, I don't have fastload (is that a Bangle JS app?) installed.

Yes, it's marked as an "experimental" app, and so when experiencing bugs the user is expected to uninstall it and see if the bug goes away before filing the bug - if they had it installed. (I've broken this rule myself sometimes...)

>Or is my understanding that Whitelist -> on should mean that the watch won't connect to anything not on the whitelist mistaken?

That is my understanding as well. [This comment](https://github.com/espruino/BangleApps/issues/78#issuecomment-666246797) might be interesting.

I don't have the answers for your other questions I'm afraid.

bobrippling, 2023-07-27:
That's interesting, there could be an issue with the whitelist handling - I have seen the address randomisation for my phone, so can't it (but I use the "Make connectable" option for attaching to it anyway).

What we could try is uninstalling Drained and seeing if the issue appears again, in which case I think we can double down on the whitelist being a potential cause. In the meanwhile, are you able to dump the relevant parts of your `setting.json` in [the IDE](https://www.espruino.com/ide/#) and I could have a go at reproducing the problem?

To do so:
```javascript
console.log(JSON.stringify(Object.fromEntries(Object.entries(require("Storage").readJSON("setting.json")).filter(kv=>/whitelist/.test(kv[0]))),0,2))
```

lightnin, 2023-07-28:
Yesterday I lost the ability to connect for a while (no drained installed), but now it's working again. Most of my attempts to make the issue reproducible / consistent have left me more perplexed, casting doubt even on my initial idea that the problem was directly "Drained" related. Bluetooth is a very hairy beast!

I'm gonna close this now with a summary of what I do know with certainty, which is not much:
It appears that after a factory reset, Whitelist in the BT menu started at "On," with no entries. But I was able to connect (eventually). This happened twice. In my view, that is very strange, as I would expect a factory reset to be the surest way of getting back to a safe / open condition, and to eliminate all state. 

That's about all the clarity I have, my friends. But I'll keep an eye out for patterns, and post if I find any that make sense!

closed 2023-07-28 by lightnin

gfwilliams, 2023-07-31:
Thanks - just to say I looked into the whitelist:on thing and fixed it - it looks like changes in settings 0.59 broke it.

However even though it showed as on, the whitelist wasn't actually on. Are you sure that when you couldn't connect, the bluetooth indicator showed greyed out? The most likely reason for connection issues is that some other device connected to the Bangle automatically (eg maybe it ended up as 'paired' to your PC, so you PC tries to auto-connect whenever it sees it)

lightnin, 2023-07-31:
Ah great! Thanks for solving this! Ok, that helps with the variable overwhelm I was suffering from. But in most all cases I did rule out something else connecting / stealing the BT connection. 

This made me wonder if the Settings > Bluetooth menu might, when connected, show the first row entry as "Connected to: a4:83 etc. etc." (or better yet, instead of a4:83 etc., show whatever human readable name is broadcast by the device it is connected to). That might be a way to communicate connected status to the user in a way that would help them identify when the BT connection is being occupied by something they weren't expecting it to be occupied by, thus preventing a connection with whatever they are trying to connect it to. I gather the widget bold / unbold status is supposed to show connectedness, but it may be a bit too subtle for some. So a little status indicator in the BT menu might help.

I will keep on the lookout for ways to isolate the issues I was seeing, and file a new report if I can make sense out of it. :) Thanks to all for looking into this with me!