espruino/BangleApps #2907: [app/btadv] Acceleromter is not working

shawsa-artc, 2023-07-25:
### Affected hardware version

Bangle 2

### Your firmware version

2v12

### The bug

App works fine except for acceleromter advertising. Seems the part accl sensor part is missing. Any plan to complete it ?

shawsa-artc, 2023-07-26:
Thanks guys, appreciated!

bobrippling, 2023-07-26:
There isn't an official bluetooth service ID for acccelerometers, but we can pretend to be [a micro:bit](https://github.com/NordicSemiconductor/bluetooth-numbers-database/blob/cecf039682185876f97be2a389e27fe04a69c475/v1/service_uuids.json#L78).

I've created #2912 but the service UUIDs seem a bit mangled when I connect to my watch, I suspect there's not enough room in the BLE advertising data for a long UUID alongside all the short ones.

As a small repro / proof-of-concept, I'm attempting to get the below code working.

```javascript
microBitAccUUID = "E95D0753251D470AA062FA1922DFA9A8";
if (!Bangle.bleAdvert) Bangle.bleAdvert = {};

pretendX = 0;
function advertise() {
	if(pretendX >= 10) pretendX = 0;
	Bangle.bleAdvert[microBitAccUUID] = [pretendX++, 0, 2, 0, 3, 0]; // 16-bit LE x, y, z
	NRF.setAdvertising(Bangle.bleAdvert);
}

setInterval(advertise, 60 * 1000);
advertise();
```

---

@shawsa-artc what device are you using that can read the accelerometer data? Do you know if it's expecting a particular service?

shawsa-artc, 2023-07-27:
> There isn't an official bluetooth service ID for acccelerometers, but we can pretend to be [a micro:bit](https://github.com/NordicSemiconductor/bluetooth-numbers-database/blob/cecf039682185876f97be2a389e27fe04a69c475/v1/service_uuids.json#L78).
> 
> I've created #2912 but the service UUIDs seem a bit mangled when I connect to my watch, I suspect there's not enough room in the BLE advertising data for a long UUID alongside all the short ones.
> 
> As a small repro / proof-of-concept, I'm attempting to get the below code working.
> 
> ```js
> microBitAccUUID = "E95D0753251D470AA062FA1922DFA9A8";
> if (!Bangle.bleAdvert) Bangle.bleAdvert = {};
> 
> pretendX = 0;
> function advertise() {
> 	if(pretendX >= 10) pretendX = 0;
> 	Bangle.bleAdvert[microBitAccUUID] = [pretendX++, 0, 2, 0, 3, 0]; // 16-bit LE x, y, z
> 	NRF.setAdvertising(Bangle.bleAdvert);
> }
> 
> setInterval(advertise, 60 * 1000);
> advertise();
> ```
> 
> @shawsa-artc what device are you using that can read the accelerometer data? Do you know if it's expecting a particular service?

I'm building a custom banglejs watch companion app using flutter right now,  any service uuid is fine as my side can be modified to subscribe to eatch service/characteristics.

bobrippling, 2023-07-27:
Sounds good - if you run the above code, are you able to pick up the accelerometer service at all?

shawsa-artc, 2023-07-28:
> Sounds good - if you run the above code, are you able to pick up the accelerometer service at all?

Hi i've tried your code, 2 problems.   

1.  service uuid E95D0753251D470AA062FA1922DFA9A8 is not picking up by the bluetooth phone app, i've modified it to use service uuid 0x1101 and char uuid 0x2101 , then no problem. It will displayed as unknow service and unknown characteristic but data is able to advertising successfully.
2.  encodeAcc_1 function is not correct, i see you are trying to encode the same way magnetic value does, but i think u should follow the encodeGps_1 instead.   Accel x y z value is showing as a long int eg '-0.099999999112321323'

gfwilliams, 2023-07-31:
I think there's a chance that there is too much data to fit into one advertising packet? It might be the 128 bit UUID ends up going into scan response advertising data.

If you need to get more data in there you can always supply an array of objects to NRF.setAdvertising, in which case it'll cycle through advertising each in turn

bobrippling, 2023-07-31:
Ok - I've got my phone picking up the long UUID now using `setAdvertising` and cycling through IDs, and the characteristic appears too. I've tweaked the encode function as well - @shawsa-artc let me know how this works for you.

shawsa-artc, 2023-08-01:
> 

the latest version works now . Thanks for the work !

3 comments on my observation.

1. accel button inside btnLayout_1 can be removed , since accel sensor is always on. there's no point putting a switch there.
2. bangle.js watch console is printing this error every second.
```javascript
in function called from system
Uncaught Error: Can't update services until BLE restart
 at line 407 col 37 in .bootcde
        NRF.updateServices(newAdvert);
``` 
3. after turn on the sensors, ble connections from bangle.js web ide/apploader will always fail while the app is on,  connections from phone has no problem.

bobrippling, 2023-08-03:
Glad to hear! I've been wanting to improve the UI for some time - while I'll sort that in a separate PR, I'd be interested in any thoughts you had on it, or any ideas from a mock-up etc.
For the error, I left that in so I'd know to reconnect if I'd left my watch attached to the app loader/IDE.
I've seen the problematic reconnect too, it's a tough one to debug but I'll see if I can save some errors to the storage for postmortem debugging.

PR #2912 (2023-08-08): btadv: sensor location & accelerometer adverts
Add accelerometer adverts and sensor location to `btadv`

Closes #2907
closed 2023-08-08 by gfwilliams