espruino/BangleApps #2913: [Bangle.js 2 Bluetooth] Unable to connect to device on latest linux kernel

0x57F, 2023-07-26:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

Unable to connect to the bangle on Linux 6.4.3-1, but after booting into Linux 5.15.120-1, I was able to connect fine. I was also able to connect in windows. This problem occurred after updating to the latest kernel version.

### Installed apps

boot (0.58), messageicons (0.05), messagelist (0.03), widmessages (0.05), messages (0.60), ios (0.13), setting (0.61), wid_a_battery_widget (0.04), accelgraph (0.01), sched (0.22), widalarm (0.01), alarm (0.41), files (0.08), mylocation (0.10), astrocalc (0.04), chess (0.02), widdst (0.04), info (0.03), longpressbuzz (0.01), matrixclock (0.07), widram (0.03), terminalclock (0.10), themesetter (0.04), dtlaunch (0.24), widpedom (0.22), widlock (0.08), notify (0.13), health (0.23), hrm (0.11), dvdbounce (0.01), tetris (0.01), bordle (0.03)

thyttan, 2023-07-26:
Works for me on Fedora Linux 38, kernel Linux 6.4.4-200.fc38.x86_64, Chromium web browser 114.

0x57F, 2023-07-27:
Interesting, I am using Manjaro 23.0.0, using chromium 115, and have tried using the latest stable version of chrome, yet the device refuses to show up in the list of available devices, and even refuses to connect when the address is specified manually using the CLI. It was not connected to anything else due to those attempts.

gfwilliams, 2023-07-27:
Does `sudo hcitool lescan` at the console show anything? If not it's definitely a Linux issue.

I'm 99% sure this is just an issue with your Linux install - maybe your bluetooth adaptor needs a firmware binary and that's not installed?

0x57F, 2023-07-27:
I get this:
```
# hcitool lescan
Set scan parameters failed: Input/output error
```
However bluetooth is still working, and using bluetoothctl in le mode, `[NEW] Device E1:AC:F3:10:27:C5 Bangle.js 27c5` does appear occasionally.

gfwilliams, 2023-07-27:
Hmm, well if you did `hcitool lescan` with su priviledges and it failed, I'd be pretty certain this is a Linux issue and not to do with Bangle.js i'm afraid.

I'm not sure we can be much help - best to ask on a Manjaro forum?

0x57F, 2023-07-27:
Will do, thanks for the support.

closed 2023-07-27 by 0x57F

thyttan, 2023-08-04:
The problem has occured for me now as well. It seems like a kernel issue: https://github.com/espruino/EspruinoAppLoaderCore/pull/48#issuecomment-1664684364

thyttan, 2023-08-04:
Relevant issue on fedora discussion forum: https://discussion.fedoraproject.org/t/no-bluetooth-found-after-upgrade-from-6-3-12-to-6-4-4/86507

gfwilliams, 2023-08-04:
Also on the forum: https://forum.espruino.com/conversations/388952

thyttan, 2023-11-08:
@0x57F Have you had any luck mitigating this problem. I've been keeping 6.3.12 and using that on fedora 38. But I just updated to fedora 39 and 6.3.12 is not stable for me anymore.

0x57F, 2023-11-09:
I think my solution was just to keep using an older kernel version - I am currently on 3.1.55-1-MANJARO which seems to work, however bear in mind that I am on Manjaro, and you are on Fedora. Sorry I could not be much more help.

thyttan, 2023-11-09:
Ok, thanks! I'll maybe go back to fedora 38 then, to be able to use the older kernel.

nxdefiant, 2023-11-09:
Same problem here with 6.1.0-13-amd64 (devuan) and a 0bda:a725 Realtek Semiconductor Corp. Bluetooth 5.1 Radio

Taking the device up and down helps for a few seconds `hciconfig hci0 reset`. Needs investigation, but definately not Bangle.js issue.

nxdefiant, 2023-12-12:
My Devuan will stay on 6.1.0 for some time..

0x57F, 2024-04-04:
Just installed the latest version of the Linux Kernel (6.8.x I think), and the issue has been resolved.