espruino/BangleApps #2915: [Compact Storage] Compact message displayed indefinitely

thyttan, 2023-07-27:
### Affected hardware version

Bangle 2

### Your firmware version

2v18.115

### The bug

I had fastload installed when this happened. I can't reproduce right now even with fastload installed.

### Describe the bug
1. I noticed quicklaunch had stopped working and went to "Rewrite Settings" which sometimes work. I tried pressing the menu entry for that but didn't get any feedback that it initialized.
2. Instead I began "Compact Storage" which displayed the old instruction on the on-screen buffer and a new instruction in a small overlay window.
3. The screen backlight timed out and the overlay window disappeared at the same time.
4. I let the watch sit for at least 10 minutes but the old instruction never went away.
5. Eventually I long pressed the HW button which took me to my clock face.
6. Navigating back to the settings app I noticed swscroll had stopped working and the firmwares E.showScroller was active instead.
7. I now managed to do "Rewrite settings" and it executed as it normally does.
8. I still didn't have my quicklaunch actions set, but swscroll now worked.
9. Went to quicklaunch settings to see that no apps were set anymore.
10. Connected to Web IDE to look at the storage file browser. There were a lot of scrambled filenames on top, like so:
[image]

### Installed apps

# Fastload 

... and others.

thyttan, 2023-07-27:
Also when trying to do a backup i get stuck with this message on the app loader:
>Download `l a

And in the Web IDE I get this when it is interrupted:
>W10=ÿUncaught Error: Cannot read property 'length' of undefined
 at line 1 col 57
...l a");Bluetooth.println(((s.length+2)/3)<<2);for(var i=0;i<s.lengt...
                                    ^
in function called from line 6 col 4
})()
   ^

thyttan, 2023-07-27:
I suspect all this will be resolved when I remove the files with scrambled filenames. That has helped before to some extent. I just wanted to document this first.

gfwilliams, 2023-07-27:
It looks like your storage has got corrupted...  Where did you start compact storage from? settings?

What do you mean by:

> I began "Compact Storage" which displayed the old instruction on the on-screen buffer and a new instruction in a small overlay window.

You mean the `Please Wait... STORAGE COMPACTION IN PROGRESS...` message? But it went away if it's own accord?

Before you do anything else, please could you try what's here? https://github.com/espruino/BangleApps/issues/2865#issuecomment-1649549028

Then I could get a copy of what's in your storage and could see if I can figure out what went wrong. Of course it may be you were having issues because storage got corrupted, and compaction just made it worse?

After that, probably the best thing is for you to do a backup of your watch using the app loader (if you can) and then delete the broken files from the zip and then restore from the zip - which will mean storage gets completely refreshed.

thyttan, 2023-07-27:
> It looks like your storage has got corrupted... Where did you start compact storage from? settings?

Yes, Settings->Utils->Compact Storage.

> What do you mean by:
> 
> > I began "Compact Storage" which displayed the old instruction on the on-screen buffer and a new instruction in a small overlay window.
> 
> You mean the `Please Wait... STORAGE COMPACTION IN PROGRESS...` message? But it went away if it's own accord?

Yes, it went away at the same time as the backlight turned off.

> Before you do anything else, please could you try what's here? [#2865 (comment)](https://github.com/espruino/BangleApps/issues/2865#issuecomment-1649549028)
>
> Then I could get a copy of what's in your storage and could see if I can figure out what went wrong. Of course it may be you were having issues because storage got corrupted, and compaction just made it worse?

I'll try to do that.

> After that, probably the best thing is for you to do a backup of your watch using the app loader (if you can) and then delete the broken files from the zip and then restore from the zip - which will mean storage gets completely refreshed.

I might do it that way then.

bobrippling, 2023-07-27:
> How long is the logging expected to take, roughly?

If it helps, I logged 251,661 lines in about two hours. I also saw shorter lines.

I was thinking, if the bottleneck's the BLE connection, we could tweak that code for next time to apply some compression.

thyttan, 2023-07-27:
> If it helps, I logged 251,661 lines in about two hours. I also saw shorter lines.

Maybe I don't see longer lines. I'm not sure.

thyttan, 2023-07-27:
570k lines now and counting.

Curiously the Web IDE connection window have popped up spontaneously a couple of times during the streaming so far. It didn't cancel the connection or stop the data stream though.

thyttan, 2023-07-27:
I just came back to the laptop/phone/bangle logging setup and the Web IDE said it had been disconnected from android. I don't know if it was that the logging was done that prompted the disconnect. At the same time it seem the bangle sent status messages saying it was fully charged. It's in the log I'll send to you @gfwilliams. Web IDE says 630,751 lines were logged.

gfwilliams, 2023-07-31:
Thanks! Well, that should be promising I hope - it's trying to write 8MB of data, and for that intel hex dump it's writing 16 bytes per line, plus some extra to set the address range - so you would expect around 550k lines I guess? 600k seems a bit high but hopefully that's everything.

We could send it as base64 I guess, which should be a bit better and faster. Intel hex is just an 'easy' format to deal with - if there is data missing from the middle it's not a big deal

thyttan, 2023-08-03:
@gfwilliams did you get the messages with my two logs on the forum? I've been holding of resetting my device in case you'd want something more from it - if there happened to be something wrong with the second log as well. But now I'm a bit eager to actually reset the watch :p

gfwilliams, 2023-08-03:
Sorry about that - yes, I just replied on the forum. Sorry for the delay - I think that's everything we can get from the watch. I'll have a go at reproducing on mine and will see if I can figure out what's up.

Interestingly it looks like there are some big erased areas in the middle, so it's almost as if it compacted, but then didn't manage to clear all the space that it had freed up. When that happens it'll still keep functioning as normal until it hits the area that it thought was clear but then isn't - and then things start to go badly wrong.

thyttan, 2023-08-03:
> Also when trying to do a backup I get stuck with this message on the app loader:
>
>
> And in the Web IDE I get this when it is interrupted:

I just tried to use the Web IDE to remove all files with scrambled names. That was possible for all but the file " `l a " referenced above.

When opening that file it's contents are `00000000: 00 00 00                                        ...
` (maybe with a newline at the end)

If I `Save` that file and open it with neovim it looks like this:
[image]
*Three identical  `^@` chars.*

And the file is this:
[`l a.txt](https://github.com/espruino/BangleApps/files/12250652/l.a.txt)

gfwilliams, 2023-08-03:
Thanks - that's interesting... Looks like the IDE hasn't been able to copy in the filename correctly.

Does doing this into the left-hand side of the IDE pane work?

```
print(JSON.stringify(require("Storage").list(),null,2))
```

You'll probably find that the filename isn't just ```"`l a.txt"``` and maybe it has some extra weird chars in there.

Then try this, but with the correct filename:

```
require("Storage").erase("`l a.txt")
```

thyttan, 2023-08-03:
> Does doing this into the left-hand side of the IDE pane work?
> 
> ```
> print(JSON.stringify(require("Storage").list(),null,2))
> ```
> 
> You'll probably find that the filename isn't just `` "`l a.txt" `` and maybe it has some extra weird chars in there.

Yes, I get this:
```
>print(JSON.stringify(require("Storage").list(),null,2))
[
  "sched.boot.js",
  "sched.js",
  "sched.img",
  "sched",
  "sched.settings.js",
  "sched.clkinfo.js",
  "sched.info",
  "messageicons",
  "messageicons.img",
  "messageicons.info",
  "widmsggrid.wid.js",
  "widmsggrid.info",
  "messagegui.img",
  "health.img",
  "widminbat.wid.js",
  "widminbat.info",
  "backswipe.info",
  "torch.app.js",
  "torch.img",
  "torch.settings.js",
  "torch.info",
  "calculator.app.js",
  "calculator.img",
  "calculator.info",
  "widram.wid.js",
  "widram.info",
  "widbthide.wid.js",
  "widbthide.info",
  "smpltmr.img",
  "stopw.clkinfo.js",
  "clkinfostopw.info",
  "bwclklite.img",
  "runplus.img",
  "dtlaunch.img",
  "quicklaunch.settings.js",
  "quicklaunch.boot.js",
  "quicklaunch.app.js",
  "quicklaunch.info",
  "alarm.img",
  "powermanager.boot.js",
  "powermanager.settings.js",
  "powermanager.wid.js",
  "powermanager",
  "powermanager.default.json",
  "powermanager.info",
  "agenda.img",
  "setting.img",
  "dtlaunch.json",
  "mysetup.info",
  "fastreset.boot.js",
  "fastreset.info",
  "health-2023-5.raw",
  "bwclklite.info",
  "smpltmr.app.js",
  "smpltmr.clkinfo.js",
  "smpltmr.info",
  "log.txt\u0001",
  "health-1970-1.raw",
[... 219 lines cut ...]
  "clock_info",
  "clock_info.info",
  "recorder.app.js",
  "recorder.img",
  "recorder.wid.js",
  "recorder.clkinfo.js",
  "recorder.info",
  "openstmap",
  "openstmap.app.js",
  "openstmap.info",
  "iconlaunch.app.js",
  "iconlaunch.settings.js",
  "iconlaunch.info",
  "sixths.app.js",
  "sixths.img",
  "sixths.info",
  ".boot0",
  "launch.cache.json",
  "fastload.cache",
  "sixths.egt\u0001",
  "fastload.history.json"
 ]
=undefined
>
```

thyttan, 2023-08-03:
> You'll probably find that the filename isn't just `` "`l a.txt" `` and maybe it has some extra weird chars in there.
> 
> Then try this, but with the correct filename:
> 
> ```
> require("Storage").erase("`l a.txt")
> ```

For clarity, the file is only named `` "`l a.txt" ``  with the `.txt` extension when it's saved to my laptops ssd.

thyttan, 2023-08-03:
Maybe [me working on Activity tracks ](https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/3153) can have had something to do with it? Just adding it as a lead.

gfwilliams, 2023-08-03:
Any luck deleting those files manually?

I'm pretty sure there was an issue with the handling of StorageFiles that could have caused problems (mainly trying to repeatedly write while storage is full). I'm not sure when you last updated your firmware but it was only very recently that I made some quite big changes to try and improve the situation, so you might have been unlucky and things got corrupted while you were on an earlier firmware?

thyttan, 2023-08-03:
I removed some more of the files using the gui and then get this:
```
print(JSON.stringify(require("Storage").list(),null,2))
[
  "sched.boot.js",
  "sched.js",
  "sched.img",
  "sched",
  "sched.settings.js",
  "sched.clkinfo.js",
  "sched.info",
  "messageicons",
  "messageicons.img",
  "messageicons.info",
  "widmsggrid.wid.js",
  "widmsggrid.info",
  "messagegui.img",
  "health.img",
  "widminbat.wid.js",
  "widminbat.info",
  "backswipe.info",
  "torch.app.js",
  "torch.img",
  "torch.settings.js",
  "torch.info",
  "calculator.app.js",
  "calculator.img",
  "calculator.info",
  "widram.wid.js",
  "widram.info",
  "widbthide.wid.js",
  "widbthide.info",
  "smpltmr.img",
  "stopw.clkinfo.js",
  "clkinfostopw.info",
  "bwclklite.img",
  "runplus.img",
  "dtlaunch.img",
  "quicklaunch.settings.js",
  "quicklaunch.boot.js",
  "quicklaunch.app.js",
  "quicklaunch.info",
  "alarm.img",
  "powermanager.boot.js",
  "powermanager.settings.js",
  "powermanager.wid.js",
  "powermanager",
  "powermanager.default.json",
  "powermanager.info",
  "agenda.img",
  "setting.img",
  "dtlaunch.json",
  "mysetup.info",
  "fastreset.boot.js",
  "fastreset.info",
  "health-2023-5.raw",
  "bwclklite.info",
  "smpltmr.app.js",
  "smpltmr.clkinfo.js",
  "smpltmr.info",
  "log.txt\u0001",
  "health-1970-1.raw",
  "hrm.app.js",
  "hrm.img",
  "hrm.info",
  "sched.settings.json",
  "lightswitch.boot.js",
  "lightswitch.images.json",
  "lightswitch.info",
  "lightswitch.js",
[... 206 lines cut ...]
  "health.info",
  "clock_info",
  "clock_info.info",
  "recorder.app.js",
  "recorder.img",
  "recorder.wid.js",
  "recorder.clkinfo.js",
  "recorder.info",
  "openstmap",
  "openstmap.app.js",
  "openstmap.info",
  "iconlaunch.app.js",
  "iconlaunch.settings.js",
  "iconlaunch.info",
  "sixths.app.js",
  "sixths.img",
  "sixths.info",
  ".boot0",
  "launch.cache.json",
  "fastload.cache",
  "sixths.egt\u0001",
  "fastload.history.json"
 ]
=undefined
>```

thyttan, 2023-08-03:
> I'm pretty sure there was an issue with the handling of StorageFiles that could have caused problems (mainly trying to repeatedly write while storage is full). I'm not sure when you last updated [...]

I think this is the likely scenario. I last worked on the PR three weeks ago, which I think is before you added the fix.

gfwilliams, 2023-08-03:
... so we're saying that you still can't back up with the app loader because you get that error?

Please can you try on the development app loader and see if it's any better? I just pushed a change to try and make readfile more robust

thyttan, 2023-08-03:
I could do the backup now thanks to your change. Then I unpacked the directory and removed the "`l a" file. Then I rezipped it and tried to restore it.

 But the `Restore` utility doesn't work. It managed to erase the device after I gave the go ahead. Then it says it reads the zip. When it says it's starting the restore it pretty immediately finishes, not enough time to have restored the zip and there's nothing on the watch.

I'll do a factory reset and then try restoring the zip without erasing first.

gfwilliams, 2023-08-03:
Ok, thanks - any errors in the console? It could be one of the files is causing an issue... You won't have all the files you need on there anyway I bet, but it should still be able to restore what it has.

Maybe email me the zip and I can see if I can figure out why it's not restoring?

thyttan, 2023-08-03:
I'll add it on the message thread on the forum. Do you want it on email as well?

gfwilliams, 2023-08-03:
No the message thread is fine - thanks!

thyttan, 2023-11-09:
@gfwilliams does it make sense to close this issue?

gfwilliams, 2023-11-10:
Yes, I'd think so... I forget but did I sort out the issue with restoring that zip file?

thyttan, 2023-11-10:
I think so, yes 🙂

closed 2023-11-10 by gfwilliams