espruino/BangleApps #2930: [alarm] Repeat setting changes from Once to 1*Month if Repeat menu canceled without changes

ticalc-travis, 2023-07-30:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

While editing an event, entering the Repeat menu while the setting is “Once”, then pressing Back to exit the menu without making a selection, results in an unexpected change of the repeat setting to once monthly. Under normal circumstances I would expect entering and immediately leaving a menu not to change any existing settings; this behavior is inconsistent with that of other menus in the interface. However, other repeat settings do not seem to be affected: Entering and backing out of the menu when a different repeat setting is active does leave the setting unchanged as expected. Only “Once” seems not to behave consistently.

Reproduction steps:
1. Open Alarms app
2. Select “New…”
3. Select “Event”
4. Scroll down to Repeat. Note default selection of Once
5. Tap Repeat menu, then immediately press Back.
6. Scroll down to Event and note the setting has changed to 1*Month. The expected result would have been for the setting to remain as Once.

I looked at the code to see if I could fix it, but I have a bit of trouble following how the repeat menu routine works exactly. I note that `showEditRepeatMenu()` gets `day` and `repeat` variables passed to it and that it seems to use a separate var `dow` to make some internal changes to `day` before the user makes a selection; but for some reason `repeat` gets manipulated directly straight away instead of a separate var as with `day`. This ultimately seems to be causing the “false” value of `repeat` to get unconditionally replaced with an object representing once monthly, and then this is getting saved back to the alarm when the user cancels the menu (despite the user not having selected any changes).

### Installed apps

launch (0.20), antonclk (0.11), about (0.14), widlock (0.08), widbat (0.11), welcome (0.14), sched (0.22), sched (0.22), widalarm (0.01), boot (0.58), notify (0.13), health (0.23), setting (0.61), alarm (0.41), mylocation (0.10), hrm (0.11), waypoints (0.01), gpsnav (0.08), sleeplog (0.15), widdevst (0.03), stlap (0.04), stlapview (0.03), batchart (0.12), dtlaunch (0.24), timerclk (0.05), widclk2 (0.00), multitimer (0.03)

Also reproducible in emulator

nxdefiant, 2023-07-30:
Can reproduce that, my guess is that the variable repeat is changed here : https://github.com/espruino/BangleApps/blob/master/apps/alarm/app.js#L301

Passing originalRepeat instead of repeat to the callback in [line 267](https://github.com/espruino/BangleApps/blob/master/apps/alarm/app.js#L267) should fix that.

ticalc-travis, 2023-08-18:
Sorry for the delay. I got around to trying your proposed change. Unfortunately, this causes another problem: The repeat can't be changed at all; it would simply be stuck on the “Once” setting. So this isn't the solution.

I think part of the root issue is the inconsistent semantics of this submenu. The “Once” option immediately sets the repeat option to one-shot and returns to the previous menu, but to change it to a repeated interval the menu expects the user to instead make the changes within the menu and then manually tap Back to save and exit. The submenu code unconditionally tries to save the settings that are made (it has no idea of “cancel”), but it also messes with the variables that go into the settings when it really shouldn't, in order for some of the menu code to work correctly.

I think fixing this properly might require a bit of reworking the code so that it doesn't need to mess with the variables representing the current settings unless the user actually selected an option. I'll have to study the code a bit more and do some thinking.

nxdefiant, 2023-08-19:
I see... well problem definitely is the switching between boolean and obj. Well we could try it the other way around: instead of saving repeat to originalRepeat create a new repeatObj
[alarm.diff.txt](https://github.com/espruino/BangleApps/files/12385207/alarm.diff.txt)

PR #3046 (2023-10-16): alarm auto expiration for events + fix repeat back
auto expiration for events:
- Set default value
- Add "delete after expiration" checkbox to events.

repeat menu: Do not change value on back (fixes #2930)
Instead of saving repeat argument as originalRepeat create a new changed repeatObj
closed 2023-10-16 by gfwilliams