espruino/BangleApps #2937: [iconlaunch] Keeps 'Loading...'

lunctis-viribus, 2023-07-31:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

After updating the `iconlaunch` launcher to v0.18 the launcher is completely broken, it hangs on the 'Loading' pop-up. I had to uninstall the launcher and install another one. I also tried uninstalling `fastload` but this didn't help either.

thyttan, 2023-07-31:
Could be the changes in this PR breaks the app on firmwares older than 2v18.115. See https://github.com/espruino/BangleApps/pull/2895#issue-1816729020.

@halemmerich for expertise.

lunctis-viribus, 2023-07-31:
Just updated to Cutting Edge build 2v18.116 but the issue is still the same unfortunately, hangs on 'Loading...'

thyttan, 2023-07-31:
It works for me on 2v18.115.

Could you copy/paste your installed app from the 'More...' tab of the app loader in a comment here?

lunctis-viribus, 2023-07-31:
Apps Installed | widid (0.03), widpedom (0.22), about (0.14), widlock (0.08), widminbt (0.01), widbt_notify (0.17), widcw (0.02), thermom (0.07), files (0.08), messageicons (0.05), locale (0.17), antonclk (0.11), weatherClock (0.06), weather (0.25), health (0.23), widmessages (0.05), widbatpc (0.19), run (0.16), clkinfomag (0.02), widhid (0.03), notify (0.13), boot (0.58), bikespeedo (0.05), messages (0.60), dragboard (0.07), gbmusic (0.12), mylocation (0.10), spotrem (0.09), android (0.29), messagegui (0.73), recorder (0.29), openstmap (0.23), setting (0.62), fastload (0.06), iconlaunch (0.18)

thyttan, 2023-07-31:
Hm, maybe try using the 'Reinstall apps' Utility under the 'More...' tab. Or the 'Reset Settings' one (clears the settings files, but shouldn't remove sensitive data I think)

lunctis-viribus, 2023-07-31:
Reinstalling all apps didn't work. I prefer not to reset all settings... I'll just use another launcher for now...

thyttan, 2023-07-31:
Check to see that 'Minify apps before upload' is toggled off. You could try toggling off 'Pretokenise apps before upload (smaller, faster apps)' as well.

[image]

lunctis-viribus, 2023-07-31:
> Check to see that 'Minify apps before upload' is toggled off. You could try toggling off 'Pretokenise apps before upload (smaller, faster apps)' as well.
> 
> [image]

They were both toggled off.

thyttan, 2023-07-31:
My last suggestion and after that I'm out of ideas. 
Install both iconlaunch and the other launcher currently working ( click 'keep both' when presented with the option in app loader). Via the launcher that works navigate to Settings->System->Launcher and select Icon Launcher. Maybe you got stuck without any launcher being selected somehow? That shouldn't make the watch behave like yours, but who knows.

bobrippling, 2023-07-31:
Would any clues be gained from loading the launcher with [the IDE](https://www.espruino.com/ide/#) attached, to observe error logs?

lunctis-viribus, 2023-07-31:
> My last suggestion and after that I'm out of ideas. Install both the launcher currently working for you and iconlaunch ( click 'keep both' when presented with the option in app loader). Via the launcher that works navigate to Settings->System->Launcher and select Icon Launcher. Maybe you got stuck without any launcher being selected somehow? That shouldn't make the watch behave like yours, but who knows.

Didn't help either, but thanks for the help.

> Would any clues be gained from loading the launcher with [the IDE](https://www.espruino.com/ide/#) attached, to observe error logs?

Good one, when pressing the button:
JIT SyntaxError: Got of expected ';'
 at line 1 col 3739 in iconlaunch.app.js
...-1,r.y+r.h-1);texted=undefined;}for(c of item){i++;let id=c.icondata||(c.iconD...

thyttan, 2023-07-31:
weird how all code seems to be on one line. 

Are you comfortable uploading directly to files in storage via the web ide? If overwrite iconlaunch.app.js with the code currently in master here: https://github.com/espruino/BangleApps/blob/master/apps/iconlaunch/app.js

nxdefiant, 2023-08-01:
`for(c of item)` doubt that this works
https://github.com/espruino/BangleApps/blob/master/apps/iconlaunch/app.js#L70

halemmerich, 2023-08-01:
It probably is the "jit" at the top of `drawItem`. Maybe support for `of` in `for` has come somewhat later than 2v18? Removing the "jit" should make it work for you. Probably removing that for now would be better if it actually doesn't (can't?) work on 2v18.

lunctis-viribus, 2023-08-01:
Can confirm this, when running the code in https://github.com/espruino/BangleApps/blob/master/apps/iconlaunch/app.js through the IDE I get the syntax error. When removing line 59 https://github.com/espruino/BangleApps/blob/master/apps/iconlaunch/app.js#L59 the `iconlaunch` app loads fine.

gfwilliams, 2023-08-03:
>  Maybe support for of in for has come somewhat later than 2v18?

`for (...of...)` has never been supported by JIT - so I'm not sure how this ever worked for you @halemmerich ? All I can think is maybe the minification actually removed the `"jit"` keyword from the function so jit was never applied for you? But I'm struggling to reproduce that here.

Although interestingly even if JIT fails I'd have hoped that it would just load the non-JIT JS and run it anyway - but maybe there's something wrong with the implementation and it bails out if JIT compilation fails.

I've just removed the `jit` keyword and pushed a fix.

closed 2023-08-03 by gfwilliams

halemmerich, 2023-08-03:
I have tested the pretokenized and decoded code directly loaded from the Bangle's storage against 2v18 and 2v18.116. It doesn't work on 2v18 as expected but it somehow does on 2v18.116 without showing the message that JIT is not working. It may well be just ignoring the `"jit";`, skipping the message and executing the function as normal JS but it still does work on my bangle. Even after factory resetting and installing `iconlaunch` 0.18 that did not change.

Some testing with really big comments in methods (in the hope JIT removes those to make a speed difference) revealed practically no difference between JITed or not when using `for .. of` or `for(;;)` and `while()`. No messages regarding not JITable functions at all. Code used here: [https://gist.github.com/halemmerich/b1f441e0e576437554b9a6035489c959](https://gist.github.com/halemmerich/b1f441e0e576437554b9a6035489c959)

gfwilliams, 2023-08-04:
Thanks! Just tried `function j() { "jit"; return 1+2 };print(j)` and it just returns the function, so JIT isn't running! Looking into it now - I bet it was a string parse optimisation I made :(

gfwilliams, 2023-08-04:
Ok, this should now be fixed! Sorry about that

halemmerich, 2023-08-04:
Great, thanks. My test code now produces expected results.
There is still some problem left with scoping in `iconlaunch`, `drawText` can not find `selectedItem`, `itemSize` and `R` variables since it will now actually be JITed and those are `const`/`let`. Giving those as arguments to the function or declaring them as `var` makes it work. Maybe `const`/`let` should be handled like `var` for the moment?

gfwilliams, 2023-08-04:
Please can you try and give me a really cut down, self-contained example of what's not working with JIT?

Is it something like this that fails for you?

```
(function() {
  var foo = 12;
  function a() { "jit"; print("foo",foo); }
  a();
})()
```

halemmerich, 2023-08-04:
What fails in `iconlauch` is this:
```
{
  const foo = 12;
  let a = function() { "jit"; print("foo",foo); };
  a();
}
```
while
```
const foo = 12;
let a = function() { "jit"; print("foo",foo); };
a();
```
and
```
{
  var foo = 12;
  let a = function() { "jit"; print("foo",foo); };
  a();
}
```
work.

gfwilliams, 2023-08-04:
Ok, thanks! I believe I've just fixed this with https://github.com/espruino/Espruino/commit/c5e1d4997a3f2121d204e650faf85f748d2c4c87 now

The above code examples now seem to work - it was a stupid omission as everything was there to make it work, it's just that the execution scope wasn't saved

halemmerich, 2023-08-04:
Very nice, these changes fix the `iconlaunch` scope problems. Does this mean scoping for `let` and `const` is complete? If so the hint in [README_JIT.md](https://github.com/espruino/Espruino/blob/0f16231a4369bb2d154b9fab5a0d806122cddb39/README_JIT.md?plain=1#L21) could be removed.

gfwilliams, 2023-08-04:
Ahh, thanks! But I'm afraid not - I think the scoping it refers to is the scoping inside a function in which they are used. They're still basically treated as `var`.

commit cd3b08e2 (2023-08-03): 0.19: Remove 'jit' keyword as 'for(..of..)' is not supported (fix #2937)