espruino/BangleApps #2943: Linux-based communication tool

pavelmachek, 2023-08-02:
I'd really like to see command line tool to communicate with watch (list/get/put files, sync time, ...). I started such tool at https://gitlab.com/tui/tui/-/blob/master/bwatch/wt?ref_type=heads but maybe someone already has better tool and it would make sense to share an effort?

thyttan, 2023-08-02:
I suspect you already know about the [Espruino Command-line tool](https://www.espruino.com/Programming#espruino-command-line-tool)?

gfwilliams, 2023-08-03:
I believe this already exists (at least for loading apps) as https://github.com/espruino/EspruinoAppLoaderCore/blob/2328a68b36a9af63c4d425f6879a62949fd69868/tools/apploader.js

```
USAGE:

apploader.js list
  - list available apps
apploader.js devices
  - list available device addresses
apploader.js install [-b1] appname [de:vi:ce:ad:dr:es]
```

So I feel like adding to that would be preferable - but as mentioned for lower level stuff, the Espruino command-line tool is probably what you need, and it already handles stuff like setting time and uploading files to storage.

closed 2023-08-03 by gfwilliams