espruino/BangleApps #2961: [Bike Speedometer] Short description of bug

smoebody, 2023-08-10:
### Affected hardware version

Bangle 2

### Your firmware version

2v18.1

### The bug

The actual speed is not shown, only three dashes:

https://www.espruino.com/ide/?codeurl=https://banglejs.com/apps/apps/bikespeedo/app.js&upload&emulator

bobrippling, 2023-08-10:
Do you have a GPS fix / is your watch able to acquire one? You could confirm this with the [gps info app](https://www.espruino.com/ide/?codeurl=https://banglejs.com/apps/apps/gpsinfo/gps-info.js&upload&emulator) - see if it displays your coordinates

smoebody, 2023-08-11:
Dear Bob,

yes I have.

The cfg.spd variable is 'true', thats why division by it equals NaN.
Happens with emulator as well.

bobrippling, 2023-08-11:
Great, cheers - would you mind confirming the fix? I've [deployed it here](https://bobrippling.github.io/BangleApps/)

gfwilliams, 2023-08-11:
Another thing to watch out for is by default the GPS doesn't send the data in the correct format (which would include speed).

Please can you make sure you upload AGPS data with the AGPS updater at https://banglejs.com/apps/?id=assistedgps - and that'll fix it

PR #2963 (2023-08-11): bikespeedo: fix locale units setting
Fixes #2961 - the original code needed exactly a number, not a boolean
closed 2023-08-11 by gfwilliams