espruino/BangleApps #2970: [lcdclock] Rendering issue in version 0.04

sculpordwarf9, 2023-08-15:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

### Describe the bug
After upgrading to v0.04, the clockface looks strange. It seems the whole clockface moves upwards and collides with the widgets:
[image]
Problem persists after I lock then unlock the screen:
[image]

### Expected behavior
Clockface looks normal after I apply the workaround below:
[image]

### Step to reproduce
1. Open Settting
2. Go back to clockface by long pressing the button.

### Workaround
Go to app launcher and then go back by touching the arrow on top left of the screen.

bobrippling, 2023-08-15:
I can't reproduce the bug here but I believe I know what the issue is. Would you mind trying out #2971?

I've deployed it to [my app pages here](https://bobrippling.github.io/BangleApps/?q=lcdclock).

gfwilliams, 2023-08-15:
Heh, looks like we were both on this. I've just fixed it :)

commit 656b8a27 (2023-08-15): 0.05: fix alignment of clock items caused by 0.04 (fix #2970)
Also move var->let, to stop fast load memory leaks
closed 2023-08-15 by gfwilliams