espruino/BangleApps #2993: ⌚️🚀 Maintainers wanted!

gfwilliams, 2023-09-01:
I've had to take a bit of time off the last few weeks with my kids and school holidays, and I haven't been as quick merging PRs as I'd like to be.

Would anyone like to get involved in checking and merging PRs? With the testing action on commits you can see pretty clearly if it's ok or not, so adding new apps is usually pretty straightforward. For changes to existing apps (especially default ones) it might sometimes need a bit of poking around and thought - not just whether the code looks ok, but thinking about whether we really want to change all the default alarms to display party parrot or whether it should be an option :)

@rigrig, @peerdavid, @nxdefiant, @bobrippling , @thyttan, @halemmerich off the top of my head I feel like you've all been contributing loads of great stuff recently, and fixing other people's code too - would you like to get involved?

And sorry if I missed anyone else from that list - please feel free to reply if you're interested!

bobrippling, 2023-09-01:
Hey, yeah I'd be interested in helping out!

gfwilliams, 2023-09-01:
That would be great! Thank you! I'll see about adding you as a collaborator

nxdefiant, 2023-09-01:
What I would like to see:
- PRs must be accepted my multiple (2? 3?) maintainers to get merged (I think github can do that)
- Maintainers can not merge/accept their own PR

thyttan, 2023-09-03:
I'd be happy to jump on board as well.

I think I align with @halemmerich's thinking. Especially with having a checklist to go through, that would help me a bunch :+1:

gfwilliams, 2023-09-04:
Thanks! I've just added @thyttan, @halemmerich and @bobrippling - I imagine there's more than enough of you for now - we may have to think of a way of stopping you from accidentally looking at the same PRs at the same time :)

@fanoush has contributed loads of quality stuff, but as you say I think it probably makes more sense for them to work on the main Espruino repository than the BangleApps one

I'm with @halemmerich on this - this is quite a small project in the scheme of things, and if I make someone admin it's because I trust their judgement. For most PRs I don't expect we'll need two people to sign off on it.

I added some initial thoughts (copy pasted from @halemmerich and added to) at https://github.com/espruino/BangleApps/wiki/App-Contribution - feel free to add to those/discuss them. I think it's probably good to be public about what kind of things we check - especially about things like when a fork is a good idea.

ProgramminCat, 2023-09-10:
I'd be interested in helping out!

gfwilliams, 2023-09-11:
@ProgramminCat thanks! I think now we've got 3 maintainers (plus me) we're probably fine for maintainers - but if we need more or someone changes their mind I'll definitely be in touch!

closed 2023-09-11 by gfwilliams

gfwilliams, 2023-11-03:
Hi @thyttan, @halemmerich and @bobrippling - thanks for spending so much time helping out BangleApps - it feels like things are going really well.

However right now the amount of GitHub emails I have to sort through every day is a bit crazy - I get an email for every comment/issue on every Espruino repo, so I've now changed my notification settings for BangleApps to `Participating and @mentions` - so I won't get emails for literally everything that comes in any more.

I guess we'll see how it goes, but it looks like you've got it sorted  - I'll try and keep up with PRs every week, but if you ever want me to look at something please just mention @gfwilliams 

Hopefully I'll actually be more likely to respond to stuff - I think recently a few things have got lost in the general torrent of emails I end up sorting through each morning :)

thanks!

thyttan, 2023-11-06:
Good to know, thanks! 🙂

bobrippling, 2023-11-12:
I was thinking about categorising the issues to see what's to be done at a glane and wanted to get some thoughts.

As I see it, issues are mostly divisible into these categories:
- Idea hatched, needs implementing
- Rough ideas, decision needed
- Uninvestigated

I was going to apply labels to the issues so we can see what might be stuck at the bottom, as opposed to the app writer (or one of us) fixing an app/lib. What do you all think?

nxdefiant, 2023-11-12:
Probably won't hurt, but you think that helps to get at least some of the issues closed? Skeptical.. unless someone just starts to work on one nothing happens.

Also from time to time I've looked through the issues, havn't found one I'm interested in.

gfwilliams, 2023-11-13:
Thanks! Yes, if you think it'd help, go for it! I think you can create new labels just fine?

Obviously there are actual bugs, but i guess we tag issues with `bug` automatically. Maybe we should tag them with `Uninvestigated` instead, and then it's easy to find the ones that haven't been looked at yet?

Someone suggested a while back there's a good-first-issue tag (https://goodfirstissue.dev/) that we could add/use to tag things that we think someone new might be interesting in picking up. I don't know if that's worth it?

Other thoughts:

* Maybe we should tag issues that are bugs with the pre-installed/generic apps/libraries as I guess those take priority over ones on individual apps
* Is `crash`/similar a good tag? I guess actual crashes/lockups take priority over little glitches
* I think we should maybe consider `idle 1+ month` as a tag - I feel like for some of these issues, the issue was reported and then one of us asked a question and the reporter couldn't even be bothered to respond. Especially if it's something that couldn't be reproduced we should probably just mark it as idle and then 1 month later close it?

nxdefiant, 2023-11-13:
+1 to close if author does not respond in a certain time, but please do not close if requirements are clear and just not fixed.

bobrippling, 2023-11-13:
Yeah, I'll see how things go - I don't want to rock the issues too much but I'll find the labels useful for searching for things to have a go at, and so on.

I've been through a couple and decided to label based on the issue's state, the area it's in (crash is a nice idea for this), and if it's an enhancement etc, whether it'd make a good first issue. I think bug/uninvestigated is a good idea too, I'll incorporate that in my next label-a-thon.

Yes - closing old issue is a good idea, but I agree, we should only do it if we can't make progress, rather than haven't

gfwilliams, 2023-11-14:
Thanks for your work on this - it really helps. Hopefully having a clearer set of issues will help potential contributors too - also, feel free to change the title of an issue if you think it isn't clear/helpful :)

gfwilliams, 2023-11-16:
That's awesome - thanks for all your work on this! Looks like you really dug into a bunch of these issues - it must have taken hours!

bobrippling, 2023-11-17:
You're welcome - it wasn't too long to be fair, gave me a good idea of old issues too