espruino/BangleApps #3000: Corrupted flash after quite heavy use

pavelmachek, 2023-09-07:
### Affected hardware version

Bangle 2

### Your firmware version

2v18

### The bug

After few weeks of rather heavy use (I was debugging Space Weaver, so writing ~1MB images), I got flash corruption. This time, I'm not aware of ever running low on battery. I was using web IDE to upload (rather small) sixths.app.js and run it, I got something about "upload failed, trying to recover", and then I got flood of:

Status 0
FW addr 0x001a2e00 fail
Status 0
FW addr 0x001a2f00 fail
Status 0
FW addr 0x001a3000 fail
Status 0
FW addr 0x001a3100 fail

...and similar. I did not catch begining of the flood and IDE does not have big-enough scrollback.

pavelmachek, 2023-09-07:
This was in Web IDE's logs:

1, \\\"sixths\\\");\\n  }\\n  if (0) {\\n    Bangle.setHRMPower(1, \\\"sixths\\\");\\n    Bangle.setGPSPower(1, \\\"sixths\\\");\\n    Bangle.on(\\\"HRM\\\", (hrm) => { bpm = hrm.bpm; } );\\n  }\\n\\n  draw();\\n  buzzTask();\\n  //accelTask();\\n\\n  if (1) {\\n    last_acc = Bangle.getAccel();\\n    aliveTask();\\n  }\\n}\\n\\ng.reset();\\nBangle.setUI({\\n    mode : \\\"clock\\\"\\n});\\nBangle.loadWidgets();\\nBangle.drawWidgets();\\nlet logfile = require(\\\"Storage\\\").open(\\\"sixths.egt\\\", \\\"a\\\");\\n\\nstart();\\n\",14336);\n\u0010\u001b[16dload(\"sixths.app.js\")\n\n"
Splitting for reset(), delay 250
XOFF received => pause upload
XOFF timeout (2000s) => resume upload anyway
>>> Sent
ERROR: [notify_error] Prompt not detected - upload failed. Trying to recover...
Splitting for Ctrl-C, delay 250
Splitting for Ctrl-C, delay 250
BT> Disconnected (gattserverdisconnected)
Disconnect callback... {"port":"Bangle.js 70d2","portName":"Bangle.js 70d2"}
WARNING: [notify_warn] Disconnected from Bangle.js 70d2
>>> Disconnected from Bangle.js 70d2
SET window.localStorage = {   "AUTO_SAVE_CODE": true,   "BAUD_RATE": 9600,   "SERIAL_IGNORE": "/dev/ttyS*|/dev/*.SOC|/dev/*.MALS",   "SERIAL_FLOW_CONTROL": true,   "SERIAL_AUDIO": "0",   "WEB_BLUETOOTH": true,   "WEB_SERIAL": true,   "EMULATOR_BANGLEJS": true,   "RELAY_KEY": "",   "RESET_BEFORE_SEND": true,   "STORE_LINE_NUMBERS": true,   "MODULE_URL": "https://www.espruino.com/modules",   "MODULE_EXTENSIONS": ".min.js|.js",   "MODULE_AS_FUNCTION": true,   "MODULE_PROXY_ENABLED": false,   "MODULE_PROXY_URL": "",   "MODULE_PROXY_PORT": "",   "ENV_ON_CONNECT": true,   "BLOCKLY_TO_JS": false,   "BLOCKLY_LANGUAGE": "en",   "BLOCKLY_EXTENSIONS": "|bluetooth|robot|",   "KEYMAP": "sublime",   "THEME": "default",   "INDENTATION_TYPE": "spaces",   "TAB_SIZE": 2,   "DISABLE_CODE_HINTS": false,   "BOARD_JSON_URL": "https://www.espruino.com/json",   "SERIAL_THROTTLE_SEND": false,   "COMPILATION": true,   "COMPILATION_URL": "https://www.espruino.com/service/compiler",   "UNICODE": false,   "MINIFICATION_LEVEL": "",   "MODULE_MINIFICATION_LEVEL": "ESPRIMA",   "MINIFICATION_Mangle": true,   "SHOW_WEBCAM_ICON": 0,   "WEBCAM_CONSTRAINTS": 0,   "FONT_SIZE": 12,   "UI_MODE": "Normal",   "SHOW_CODE_LINK_ICON": false,   "ENABLE_Testing": false,   "Sound_Success": "",   "Sound_Warning": "",   "Sound_Error": "",   "Sound_Info": "",   "Speak_Success": false,   "Speak_Warning": false,   "Speak_Error": false,   "Speak_Info": false,   "SHOW_TERMINAL_LOGGER_ICON": false,   "TERMINAL_LOGGER_SAVE_TO_LOCALSTORAGE": false,   "PRETOKENISE": false,   "SAVE_ON_SEND": 0,   "SAVE_STORAGE_FILE": "sixths.app.js",   "LOAD_STORAGE_FILE": 2,   "SET_TIME_ON_WRITE": true,   "WEBRTC_BRIDGE_ID": "",   "NOT_FIRST_RUN": true }
SET window.localStorage = {   "AUTO_SAVE_CODE": true,   "BAUD_RATE": 9600,   "SERIAL_IGNORE": "/dev/ttyS*|/dev/*.SOC|/dev/*.MALS",   "SERIAL_FLOW_CONTROL": true,   "SERIAL_AUDIO": "0",   "WEB_BLUETOOTH": true,   "WEB_SERIAL": true,   "EMULATOR_BANGLEJS": true,   "RELAY_KEY": "",   "RESET_BEFORE_SEND": true,   "STORE_LINE_NUMBERS": true,   "MODULE_URL": "https://www.espruino.com/modules",   "MODULE_EXTENSIONS": ".min.js|.js",   "MODULE_AS_FUNCTION": true,   "MODULE_PROXY_ENABLED": false,   "MODULE_PROXY_URL": "",   "MODULE_PROXY_PORT": "",   "ENV_ON_CONNECT": true,   "BLOCKLY_TO_JS": false,   "BLOCKLY_LANGUAGE": "en",   "BLOCKLY_EXTENSIONS": "|bluetooth|robot|",   "KEYMAP": "sublime",   "THEME": "default",   "INDENTATION_TYPE": "spaces",   "TAB_SIZE": 2,   "DISABLE_CODE_HINTS": false,   "BOARD_JSON_URL": "https://www.espruino.com/json",   "SERIAL_THROTTLE_SEND": false,   "COMPILATION": true,   "COMPILATION_URL": "https://www.espruino.com/service/compiler",   "UNICODE": false,   "MINIFICATION_LEVEL": "",   "MODULE_MINIFICATION_LEVEL": "ESPRIMA",   "MINIFICATION_Mangle": true,   "SHOW_WEBCAM_ICON": 0,   "WEBCAM_CONSTRAINTS": 0,   "FONT_SIZE": 12,   "UI_MODE": "Normal",   "SHOW_CODE_LINK_ICON": false,   "ENABLE_Testing": false,   "Sound_Success": "",   "Sound_Warning": "",   "Sound_Error": "",   "Sound_Info": "",   "Speak_Success": false,   "Speak_Warning": false,   "Speak_Error": false,   "Speak_Info": false,   "SHOW_TERMINAL_LOGGER_ICON": false,   "TERMINAL_LOGGER_SAVE_TO_LOCALSTORAGE": false,   "PRETOKENISE": false,   "SAVE_ON_SEND": 0,   "SAVE_STORAGE_FILE": "",   "LOAD_STORAGE_FILE": 2,   "SET_TIME_ON_WRITE": true,   "WEBRTC_BRIDGE_ID": "",   "NOT_FIRST_RUN": true }

pavelmachek, 2023-09-07:
Watch is stuck in "BangleJS 2v18" screen with copyright. (I'd not mind bigger font there). IDE can't connect. If I hold the button, it adds "Loading..." and is stuck there. If I hold button for some more, I get "Checking storage... Storage is corrupt." and I'm getting intro screen. Time to pull the backup, I'm afraid.

pavelmachek, 2023-09-07:
Hmm. Yesterday I installed "adjust clock" widget.  I understand it writes to filesystem a lot and in strange times. Could it be related?

gfwilliams, 2023-09-07:
Hi - yes, I think the best bet is to apply a backup.

If you have stuff on there you want to keep that isn't backed up, you could try *making* a backup, deleting any obviously corrupted filenames from the zip, then erasing to factory settings and 'restoring' from the zip (without the 'delete all' option).

Is it likely you were doing anything with StorageFile (not just Storage - the one that uses `require("Storage").open(...)`). That's been the thing that seemed like the biggest culprit lately, and I did make some changes that are in the cutting edge builds to try and fix that

pavelmachek, 2023-09-07:
Yes, I have -- let logfile = require("Storage").open("sixths.egt", "a"); -- and then write to it a bit, so that's possible culprit.

gfwilliams, 2023-09-07:
Ahh - interesting. I bet that'll be it then.

If you move to the cutting edge build hopefully that'll fix it for you!

bobrippling, 2023-11-15:
I think this is a duplicate of #2865 - please reopen (there) if you see this again though

closed 2023-11-15 by bobrippling