espruino/BangleApps #3003: App Multiloading Issue

Catking473, 2023-09-09:
### Affected hardware version

Bangle 1

### Your firmware version

2v19

### The bug

Bug Description
When I open the apps Astroids and Car Crazy, they will glitch out. It is half the game screen, then it flashes and renders the launcher and it is very broken. After spamming the middle button to open the apps, it appears to open multiple times. The car crazy app would be very laggy and it was very broken, while the astroids app is fine until you shoot an astroid and hit it. I'm pretty sure its a software issue. I've tried resetting the settings and apps to default, but it doesn't change anything. This occurs on the normal launcher and the desktop launcher. I haven't tried others.

What I Expected
Astroids and Car Crazy to not glitch out on opening. I don't think this is an app issue

Steps to reproduce
Get BangleJS 1 on software 2v19, reset to default settings, install astroids/car crazy app or both and attempt to run them.

### Installed apps

Bangle.js 1 default apps from banglejs.com/apps
latest version avaliable on 9/9/2023
(I resetted my apps so thats why it doesn't have the default format here)

gfwilliams, 2023-09-11:
> After spamming the middle button to open the apps

I'm not sure I understand what you mean here? Just being in the launcher and pressing the middle button lots of times? But it only happens with Car Crazy/Asteroids?

Someone complained about Asteroids and I did see that same happening with it when I looked a week or two ago. I believe it's to do with the use of `Bangle.setLCDMode("doublebuffered");` which puts the Bangle.js 1 LCD into a narrow double-buffered mode, but I haven't yet been able to figure out what exactly is broken because when used on its own it seems to work.

bobrippling, 2023-09-11:
I've had a local app I'm working on that could be seeing a similar issue.
It's with `setLCDOverlay()` but I notice occasionally the foreground colour on the overlay changes to an olive green - I'm not redrawing on the overlay and unfortunately `g.dump()` doesn't catch the overlay, but I'll see about getting the app to a reasonable state if you think it could be related / help track down an LCD issue?

bobrippling, 2023-09-15:
> Just fixed that issue though - looks like a 1bpp overlay used the current FG/BG color like it would if you rendered a 1bpp image

Ah great, thank you! I'll try that out or jump onto the new firmware

gfwilliams, 2023-10-16:
I've just looked into this a bit more and I found out what the problem was - strangely it was related to the terminal!

I believe you'll find this is all fine if you install the latest cutting-edge firmware (or 2v20 when released)

closed 2023-10-16 by gfwilliams

blueiicey, 2023-10-16:
This fixed the issue completly! Thanks so much! Now I can play Astroids and Car Crazy!