espruino/BangleApps #3009: [string].charCodeAt behaves differently in fw 2.19 compared to fw 2.18 and could cause problems in not updated apps / games

joyrider3774, 2023-09-11:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

### bug
I had updated my firmware from 2v18 to 2v19 on my bangle and my game i was working on was no longer working. It turns out it ended up in an endless loop because of a change in behaviour with the `[string].charCodeAt` function. In firmware 2v18 `[string].charCodeAt(index)` would return 0 if you used an index value that was higher than the strings length. In firmware 2v19 it now returns -1 because my game relied on the return value of 0 it became stuck in an endless loop and potentially if other apps rely on the same behaviour from fw 2v18 they could become stuck in an endless loop as well.

I don't know what the correct behaviour should be but looking up the "javascript" documentation for it and testing in chrome this for example produces NaN `'a'.charCodeAt(2)`  while as said before the same code in fw 2v18 returned 0 and in fw 2v19 now -1.

I don't know what it should correctly return NaN maybe as to stay the same as what happens in browsers ? But i thought i would put up an issue if only as a warning for other people to see

### Installed apps

launch (0.20), antonclk (0.11), about (0.14), welcome (0.14), sched (0.22), sched (0.22), widalarm (0.01), boot (0.59), notify (0.13), health (0.27), setting (0.63), alarm (0.42), widchime (0.02), hrm (0.11), messageicons (0.05), messagegui (0.73), widmessages (0.05), messages (0.61), android (0.31), agenda (0.14), acmaze (0.03), altimeter (0.03), info (0.03), magnav (0.06), sleeplog (0.15), hwid_a_battery_widget (0.10), locale (0.17), widstep (0.01), dtlaunch (0.24), widdevst (0.03), widlock (0.08), widhr (0.01), waternet (0.01)

bobrippling, 2023-09-13:
Yeah, `NaN` is what the [spec defines](https://262.ecma-international.org/14.0/#sec-string.prototype.charcodeat), I imagine we could change it to that if there's not a big performance issue, although according to [this comment in the source](https://github.com/espruino/Espruino/blob/d8d3844e9d2566a36c20e49b4e8add9fa54b355f/src/jswrap_string.c#L143-L143), zero may be intentional.

-1 is only returned if `charCodeAt` is called on a non-string - could it be that the game you're working on calls it on something else?

gfwilliams, 2023-09-13:
I've just fixed this - so cutting edge builds or 2v20 (when released) will have the change in

But I guess for your code you're probably best trying to work on reported String length now - because depending on the version charCodeAt will return 0/-1 or NaN

closed 2023-09-13 by gfwilliams