espruino/BangleApps #3010: gipy: Link to issues in Readme is broken

nxdefiant, 2023-09-12:
Original statement is:

`[https://github.com/wagnerf42/BangleApps](here)`

This results is a link "https://banglejs.com/apps/apps/gipy/here" with the text "https://github.com/wagnerf42/BangleApps" 

Please note that https://github.com/wagnerf42/BangleApps does not have issues enabled.

@wagnerf42

nxdefiant, 2023-09-12:
And as Bangle.js app developers, do "we" actually want to have bugs reported in private repositories?

wagnerf42, 2023-09-12:
i've enabled the issues but i think it's actually better to have them here where other people might find it.

gfwilliams, 2023-09-13:
> i think it's actually better to have them here where other people might find it.

Yes, absolutely! I believe it is possible to format a link for a new issue such that it assigns someone to the issue if you do want to be notified of issues: https://docs.github.com/en/issues/tracking-your-work-with-issues/creating-an-issue#creating-an-issue-from-a-url-query

PR #3015 (2023-09-13): gipy: fix links and trim whitespace
@wagnerf42 coincidentally I used gipy the other morning - great app! This PR fixes the links in the readme (no need for a version bump since there's no code changes)

Closes #3010
closed 2023-09-13 by gfwilliams