espruino/BangleApps #3014: Feature / Typescript Golf

ddanon, 2023-09-12:
# Description

Rewriting the golf score keeping app in Typescript

## Why?

Mostly for my own learning and enjoyment! I would like to add an app related to the golf score keeping app that simply tracks :disappointed: :neutral_face: or :smile: for each hole. (I golf to get steps in; this kind of fun not-score keeping is more important to me than actual strokes taken.)

## Who?

I plan to work on this myself for fun. If someone else wants to help out, awesome! If not, that's totally fine.

jdekarske, 2023-09-12:
I plan on updating my [golf app](https://github.com/espruino/BangleApps/tree/master/apps/golfview) this offseason. Reach out if you'd like a tester! :smile:

ddanon, 2023-09-24:
@jdekarske thank you! I definitely will once I get some time!