espruino/BangleApps #3020: [app/messagegui] ??? instead of unicode text

ksavinov, 2023-09-17:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

Text inside message: signs "?" instead of unicode symbols.

[image]

### Installed apps

boot (0.59), launch (0.20), antonclk (0.11), notify (0.13), health (0.27), setting (0.63), about (0.14), sched (0.22), widalarm (0.01), alarm (0.42), widbat (0.11), widbt (0.09), widlock (0.08), widid (0.03), messageicons (0.05), messagegui (0.73), widmessages (0.05), messages (0.60), android (0.31)

gfwilliams, 2023-09-18:
Are you using this with Gadgetbridge?

Please can you follow the steps under https://www.espruino.com/Gadgetbridge#extra-setup - specifically you want to enable `Text as Bitmaps`

thyttan, 2023-09-18:
> No. Gadgetbridge is not in my app list, I installed 'Android Integration' app instead, because of description in Gadgetbridge:
> `(NOT RECOMMENDED) ... Please use the 'Android Integration' Bangle.js app instead.`

I'm pretty sure @gfwilliams meant if you use the Gadgetbridge app on your android device - not the one on the app loader 🙂

gfwilliams, 2023-09-18:
> I'm pretty sure @gfwilliams wondered if you use the Gadgetbridge app on your android device - not the one on the app loader 

Yes... I wonder whether I should just remove it. 2% of users do apparently still have it installed though, which still seems like quite a lot

gfwilliams, 2023-09-18:
I can pretty much guarantee that nothing will happen and then someone will complain when we do take it out.

How about we add deprecated:true to the JSON, and then a `Show Deprecated App` option to the App Loader and hide it under `Advanced Settings`?

ksavinov, 2023-09-18:
Concerning the current issue, I can confirm that after switching on `Text as Bitmaps` parameter - issue is not reproduced anymore. So, looks like my bad 🙂

However, I still have formatting problems (please, see the following screenshot).
And some of new messages (not all of them, can not reproduce it stably) disappears.
Probably these are another issues. Can we close the current issue?
[image]

thyttan, 2023-09-18:
>How about we add deprecated:true to the JSON, and then a Show Deprecated App option to the App Loader and hide it under Advanced Settings?

Sounds good to me! But I guess there's no hurry.

gfwilliams, 2023-09-18:
Ok, glad that sorted it! I'll close this now.

But yes, unfortunately when Bangle.js doesn't have the required fonts it has to render on the phone. But when it's rendered on the phone it doesn't know what size font to use so everything ends up the same size. That is configurable though (in Gadgetbridge on Android) so you could make the text a little smaller there (and set the 'minimum font size' in the Settings->Apps->Messages on the Bangle) to avoid having too many problems.

We are actually getting some way towards fixing that properly though - the 2v19 firmware comes with the ability to render Unicode from font files, so hopefully soon we'll be able to render characters on the watch and avoid a lot of these issues

I'd have a check as there may be issues open on some of that already though.