espruino/BangleApps #3026: Watch stucks on "Loading..." after long waiting

ksavinov, 2023-09-22:
### Affected hardware version

Bangle 2

### Your firmware version

2v19

### The bug

Previously I installed several new apps. I thought that any of new app is the reason of the issue. I suspected `cassioWatch` , but after the rallback to the the previous clock app - the problem still reproduces permanently.

After several hours of waiting display stucks with "Loading..." message, with no further actions
[image]

While rebooting the watch, I also noticed `BUFFER_FULL` message:
[image]

After reboot I still get the same behavior.

There's still a free memory on the device:
[image]

For now I don't understan if it's app related issue. And if it's app related issue, how can I find out what app affects this behavior (maybe there are tools for that?)?
Is there a workaround for this issue?

### Installed apps

dragboard (0.08), android (0.31), messages (0.60), messagegui (0.73), widbat (0.11), alarm (0.42), boot (0.59), notify (0.13), health (0.27), setting (0.63), about (0.14), sched (0.22), widalarm (0.01), widbt (0.09), widlock (0.08), widid (0.03), messageicons (0.05), widmessages (0.05), showimg (0.02), fileman (0.03), dtlaunch (0.24), weather (0.25), owmweather (0.03), dvdbounce (0.01), compass (0.09), espruinoterm (0.01), waypoints (0.01), speedalt (0.13), gbmusic (0.12), antonclk (0.11)

gfwilliams, 2023-09-25:
Please can you go to https://www.espruino.com/ide/ and connect to the watch (top left button, web bluetooth), and then try and change apps, and copy-paste anything that is displayed in the left-hand side of the screen here? Hopefully an error message will be shown that will tell us what's wrong.

But you could also try going to https://banglejs.com/apps/ and using the re-install apps button under 'More'. That won't mess with any settings but it will re-write all the code for all apps in case something happened that would cause a problem.

ksavinov, 2023-10-02:
Sorry for a late answer, I had some problems with connections IDE to Bangle.js. So, I decided to re-install all apps in my watch.

First of all, in Gadgetbridge I selected `More...` > `Reset Settings` (it didn't help to solve my problem). Then I selected `More...` > `Remove All Apps` (it helped to solve my problem). Then I started to re-install previously installed apps. "Loading..." issue reproduced when I installed `gbmusic (0.12)` app. My issue was solved with removing `gbmusic (0.12)`. Probably this application was the reason of my problem. Thank you for your help!

gfwilliams, 2023-10-03:
Ok, thanks! Do you have any other settings set in the app loader under `More...` (like `Translations` or `Minify apps before upload`?)

I've just tried installing `gbmusic` here and it doesn't seem to stop the watch loading properly, so it's a bit hard to me to know what's wrong. Maybe when it's broken you could do `Backup` from the app loader and attach the zip file it creates? That way I can reproduce it here and try and see what's wrong

ksavinov, 2023-10-03:
As I could see, settings `Translations` or `Minify apps before upload` are inactive.

I reproduced issue with `gbmusic`. Problem reproduced after I reinstalled this app and waited for about a couple of hours.
However, I could not download backup (maybe I'm doing something wrong). My watch and Gadgetbridge stucked like shown on the following screenshots. I waited for a bit, but both watch and Gadgetbridge are still in the same state.

[image]

[image]

bobrippling, 2023-10-03:
I understand that connecting [the IDE](https://www.espruino.com/ide/) hasn't worked in the past - perhaps we could retry this approach to see if there's any error logs on your bangle?

What problems did you have connecting the IDE to your Bangle before?

gfwilliams, 2023-10-04:
I didn't realise you were connected by Gadgetbridge! That makes a big difference.

I think the backup could be done from outside Gadgetbridge (disconnecting Gadgetbridge, and going to https://banglejs.com/apps on your device instead). **However there probably isn't a need** as it's probably something Gadgetbridge is sending to the watch that is causing problems.

Please could you turn on logging in Gadgetbridge, and then send me Gadgetbridge's log file the next time this happens to the watch? It would contain the contents of any notifications you received so you might want to send it just to me personally at gw@pur3.co.uk

Realistically that's the only way I can track this down - it's not something that 'just happens' by leaving it - it'll be something (maybe a notification) that Gadgetbridge sends to the watch that is causing `gbmusic` to fail

ksavinov, 2023-10-05:
Sure, thanks for these steps. I'll do it next time when I reproduce the stuck issue.

Concerning the problems with connecting IDE to Bangle, I could not connect my watches. I looked through the forum topics about probable problems, and will try to troubleshoot more carefully.